hneemann/Digital #228: Audio output?

mitolegy, 2019-01-10:
Hi, I just wanted to ask if it would be possible to add a speaker element to digital. Logisim had a mod that added one such element and it would allow you to control the frequency of the sound it generated through an input on the side. the only problem is that logisim is so slow that once you start getting system powerful enough to create something interesting the clock is running at like 100hz. 

digital seems much more capable and a speaker element could be a really neat new output method.

hneemann, 2019-01-11:
Do you relay want to set the frequency? In this case you can only generate sine waves.
If you want to do something interesting, you have to set the amplitude. 
And in this case also Digital is to slow to create signals with a frequency above some kHz.

mitolegy, 2019-01-11:
maybe you could select the amplitude and what waveform it would generate in the components properties menu. then just map the frequency it generates directly to the input value aka an input value of 256 will generate a waveform with the frequency of 256, a value of 1024 will make one at 1024 Hz etc. most early computers just had a few voices that could only generate one waveform at a time anyway so you can make some pretty neat things even just with that. I also think one of the mod components for logisim had a separate input port for amplitude. allowing you to change the volume.

hneemann, 2019-02-10:
@mitolegy:
Please try the latest [pre release v0.21-73-gc6bf863](https://infdigital.dhbw-mosbach.de/) build.
I have added a MIDI component. There is a MIDI example in the examples/misc folder.
Please let me know what you think.

mitolegy, 2019-02-11:
> @mitolegy:
> Please try the [pre release v0.21-73-gc6bf863](https://infdigital.dhbw-mosbach.de/) build.
> I have added a MIDI component. There is a MIDI example in the examples/misc folder.
> Please let me know what you think.

Hi!
I think the component is great. I did notice that it won't allow me to select MIDI channel 16 tho, other than that I think its a nice addition. 
I do have one question tho.
I know the component is in the very early stages of development but will it be possible for users to add their own custom midi instruments for the component to use?

hneemann, 2019-02-11:
@mitolegy 
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/download?file=data/Digital.zip122286840).
There is also a new funny example.

mitolegy, 2019-02-12:
the newest version is very impressive. with so many sounds to choose from, I can't wait to see what people will make. And the example you added is a very good demonstration of what it's capable of. I think this will be a very nice addition to the IO section.

commit ae070ddd (2019-02-15): adds a MIDI component; closes #228
closed 2019-02-15 by hneemann