hneemann/Digital #231: Feature request: partial write memory(32 bits)

itviewer, 2019-01-15:
Logisim is very popular in computer architecture teaching, especially UC Berkeley's [cs61c](http://inst.eecs.berkeley.edu/~cs61c/fa18/) course, which textbooks is based on _Patterson &	Hennessey,Computer Organization and	Design_. Although the Logisim project itself has stopped, the Logisim-Evolution project adds a lot of functionality to the original, including partial write support for memory, which is very convenient for cpu architecture research and simulation,such as for **sb, sh, sw** instructions.

In Digital, we can achieve this through the chip select memory combination, but  this is inconvenient and it is difficult to load the initial data to memory.  

I tried to modify the RAMDualPort component by adding a 2-bit input port(called mask and 00 as sb, 01 as sh, 10 as sw instruction). [RAMDualPortWriteMask.txt](https://github.com/hneemann/Digital/files/2758123/RAMDualPortWriteMask.txt) , there are very few changes, so there are a lot of duplicate code with the RAMDualPort component.  I want to implement this by inheriting the RAMDualPort class, but since many properties are private, It seems need to add functions and modify the `readInputs` method. also add new component need to add some language file, 

As it need to modify origin code,  and I am not familiar with German and English, so I think  if you agrees and would like to do this, it is a better way for the author to implement it.

hneemann, 2019-01-15:
Please let me know, how this new RAM components works in your use case.

closed 2019-01-15 by hneemann

itviewer, 2019-01-20:
> Please let me know, how this new RAM components works in your use case.

Sorry for replying so late, I am still stuck in how to implement  access with byte-addressing but 32-bit data width memory, so that sb, sh, sw and lb, lbu, lh, lhu, lw instructions(risc-v) can be implemented. I am still exploring

commit 9816cae4 (2019-01-15): adds dual-port multi-byte memory that allows single bytes to be overwritten; closes #231