hneemann/Digital #233: No built-in memory component for byte-addressing but multi-bit data width access

itviewer, 2019-01-21:
Suppose there is such a use case, 
1、store immediate data `0x44332211` into the fifth register (`x5`),
2、store the contents of the fifth register(`x5`) to memory address `0x12`, 
3、then fetch halfword from memory address `0x13` to register `x6`.

for risc-v instructions
```
li x5,0x44332211
sw x5,0x12(x0)
lh x6,0x13(x0)
```
we can use [venus](https://github.com/ThaumicMekanism/venus) to simulate

[image]
The right side is the memory content (the li pseudo instruction is translated into two instructions). The first four words correspond to the instructions code. 

For data access, since we don't need alignment, we can store and access it arbitrarily. we store `0x44332211` to memory by instruction `sw x5,0x12(x0)` and we can load data `0x3322` from address 0x13. 

It seems that it cannot be achieved by the [commit (https://github.com/hneemann/Digital/commit/9816cae47124b8c841bf34a23ab032cb7b1ec4a0) , maybe we can use four 8-bit memory component with chip selection feature to achieve this, but I am not sure how complicated.

hneemann, 2019-01-21:
You're describing an unaligned 32-bit memory access. This is impossible to implement with a 32-bit memory device because there is only one address input available. If you want to support unaligned memory access, the only way to do this in hardware is to use four 8-bit memory components that allow you to use different addresses for each byte.
Of course, it would be possible to add a special RAM chip that contains all the necessary logic, but that feels strange to me. Almost a bit like adding a complete RISC-V processor as a chip. I have to think about it.

hneemann, 2019-01-21:
First, I would check if a misaligned memory access is at all needed. I suspect that in most compilers you can simply disable this. Next, I would check if an interrupt controller should be implemented. If an interrupt controller is available anyway, it would also be an option to implement these memory accesses (And it would be a nice exercise too :-).
And to be honest, I'm not an FPGA expert. I can't say right away what to consider when using block RAM if such accesses are to be implemented as single cycle accesses. I could imagine that this is not much easier than an implementation with separate 8-bit RAMs in Digital.

hneemann, 2019-01-22:
Here is an exemplary circuit, which is able to process also not aligned memory accesses. 
To achieve this, a lot of complex logic is required.

[ramUnaligned.dig.zip](https://github.com/hneemann/Digital/files/2782932/ramUnaligned.dig.zip)

TheRealOrangus, 2019-01-23:
This can also be done using just two 32bit-word RAMs.
[Unaligned Adaptor.zip](https://github.com/hneemann/Digital/files/2785814/Unaligned.Adaptor.zip)

closed 2019-01-24 by hneemann