hneemann/Digital #237: Brazilian Portuguese translation

crcpucmg, 2019-02-01:
I am sending attached files for the first version of the Brazilian Portuguese translation.
I'm having found some difficulty in adjusting the lengths of the mouse tool tips, as well as finding some newlines where there are only common spaces. Shortening the text did not have the desired effect. Any suggestions on what I should do? 
P.S .: Some typos in the English language xml file have also been edited.
[lang_pt_br.zip](https://github.com/hneemann/Digital/files/2823606/lang_pt_br.zip)

hneemann, 2019-02-05:
Thank you very much for your efforts.
I added the translation to the repo. A few keys are still missing.
This way you can create a list of missing keys:
```
mvn -Dtest=TestLang test
```
It would be nice if you could add them.

crcpucmg, 2019-02-05:
Hi,

I still can not do the missing key tests.
I'm currently experiencing some problems with Maven's installation of the
project, especially with surefire.
It may be a version problem, but I have not been able to get around it yet.

In the meantime, the translated file will be reviewed, and the missing
keys will be edited.
Some terms I chose to keep in English, as they are most commonly referred
to.

I'll keep in touch.

T.

hneemann, 2019-02-06:
I'm using maven 3.3.9 and on travis-ci maven 3.5.2 is used.
You can also find the list of missing keys in the [travis-ci logs](https://travis-ci.org/hneemann/Digital) (line 2533).
When you create a fork and a pull-request, travis-ci also builds your pull request and you are able to inspect the results of the language related tests.

crcpucmg, 2019-02-10:
Thank you for the information you sent (mainly travis-ci.logs).

The translations of the new keys have been added.
A new file was sent as a new issue.

Please inform us if more is required.

hneemann, 2019-02-10:
See #242

closed 2019-02-10 by hneemann