hneemann/Digital #240: Input/Output pin order in Simple shape changes from 0.19 to 0.20+

dfbg, 2019-02-07:
Opening some circuits in newer versions I discover they won't run.  It appears two things have changed.

The sub-circuits I've set to the "Simple" shape have reverted to "Default" with the IC layout that moves the pin locations and breaks the connections.  This is easy (if tedious) to fix by switching the shape back.

But it has also rearranged the order of both the input and output sides in "Simple" mode.  Which is much more work to fix.  And it's also happened to library parts like 74138 (e.g. changing the select pins from ABC to ACB).  I don't want to edit the order of any of these circuits but I really don't want to edit those.

hneemann, 2019-02-08:
> The sub-circuits I've set to the "Simple" shape have reverted to "Default" with the IC layout that moves the pin locations and breaks the connections. This is easy (if tedious) to fix by switching the shape back.

You can use the "group edit" function. Select the complete circuit and click right to modify all chips at once.

> But it has also rearranged the order of both the input and output sides in "Simple" mode. Which is much more work to fix. And it's also happened to library parts like 74138 (e.g. changing the select [...]

The last change of pin positions in 74138 (unintentionally by fixing #150, it was not possible to select other than a DIL shape at that time) was before 0.19 was released. Are you sure that in 0.19 it behaves like expected? Or did you use 0.18 while building your circuits?

hneemann, 2019-02-08:
I looked a little deeper into the git history: In my previous message I was wrong: The pin order of 74138 never changed. It was always ACB.
Could you please attach a broken example so I can figure out what's going on?

hneemann, 2019-02-15:
@dfbg 
If the problem still exists, please feel free to reopen the ticket.

closed 2019-02-15 by hneemann