hneemann/Digital #242: Brazilian Portuguese Translation update (v1)

crcpucmg, 2019-02-10:
Brazilian Portuguese Translation updated,
some minor edition and new keys added.

[lang_pt_update_v1.zip](https://github.com/hneemann/Digital/files/2849342/lang_pt_update_v1.zip)

closed 2019-02-10 by hneemann

hneemann, 2019-02-10:
@crcpucmg 
You can find the complete translation in the latest [pre release build](https://infdigital.dhbw-mosbach.de/).

You may have noticed that the PDF documentation is automatically generated from the lang_xx.xml files. In this step, a static text part is added to the documentation. So far this part is missing in the Portuguese documentation. This static part can be found in the file [src/test/resources/docu/static_en.xml](https://github.com/hneemann/Digital/blob/master/src/test/resources/docu/static_en.xml). 
Are you interested in translating the chapters "First Steps" and "Hierarchical Design" to Portuguese as well? In this case I would create the necessary screenshots.

crcpucmg, 2019-02-12:
I'll take a look, and if there's any free time, you can count on me.

crcpucmg, 2019-02-13:
The translated file static_pt.xml was uploaded as a new issue.

commit 3a1951ae (2019-02-10): completed Portuguese translation; closes #242