hneemann/Digital #244: javaw.exe process does not exit after closing the application under Windows7

itviewer, 2019-02-13:
on Windows, since the` javaw.exe` process does not exit, the next time you open the app, the status bar prompt "The remote port is in use! Is there an other instance running?"
my environment is windows7 with

> java version "1.8.0_202"
Java(TM) SE Runtime Environment (build 1.8.0_202-b08)
Java HotSpot(TM) 64-Bit Server VM (build 25.202-b08, mixed mode)

It does not occur under Ubuntu.

hneemann, 2019-02-13:
Do you have to do something special or it is possible to reproduce that behavior by simply open Digital and close it again?

itviewer, 2019-02-14:
I don't think I have done anything special.  do you have a Windows system to test it?

hneemann, 2019-02-14:
That's the problem: I use the simulator a lot on a Win7 machine with Java 1.8 installed and never observed such a behavior.

itviewer, 2019-02-14:
It's strange.
Is there any code that will exit the application after the last visible window close in the Main.java?
as we set `setDefaultCloseOperation(DO_NOTHING_ON_CLOSE);`, when we close the window, I think it just hide it not really close.
maybe we should add  some code in the program, so when the last visible window is closed, we should quit the program by `System.exit(0)` explicitly;.

hneemann, 2019-02-14:
There is no special code necessary. If the last thread which is not marked as a demon thread terminates, also the JVM terminates. Setting a frame to DO_NOTHING_ON_CLOSE is required to deal with the "save changes" dialogs. These dialogs disposes the frame if the user selects "quit". You can find the code in the ClosingWindowListener.
There must be a thread which is still alive or a window which is not disposed which keeps the Swing thread alive. The question is now: Which thread is still running after you have closed the last Digital window.

hneemann, 2019-02-14:
Are you familiar with jConsole? jConsole is part of the JDK and you can find it in the JDK/bin folder.
Please start Digital, than start jConsole and connect it to the Digital JVM. Now you can select the "Threads" tab to see the threads which are running.  If now Digital is closed you can see, which threads are still running.

itviewer, 2019-02-15:
I use another windows10 computer for test, it does not have this problem. what's your OS version?
If you use windows7, it should be my computer environment problem. Otherwise it maybe a java bug on windows7. anyway, I can ignore this problem.

hneemann, 2019-02-15:
Just for clearance: 
You start Digital by starting Digital.exe, than you close the Digital window, not doing anything else. And than the javaw.exe remains running?

itviewer, 2019-02-15:
yes,no matter use Digital.exe or double click the jar file directly. not do anything.

hneemann, 2019-02-16:
Could you please replace the DO_NOTHING_ON_CLOSE default close operation in the Main class by DISPOSE_ON_CLOSE? It's interesting to me what's happening in this case.

itviewer, 2019-02-17:
> Could you please replace the DO_NOTHING_ON_CLOSE default close operation in the Main class by DISPOSE_ON_CLOSE? It's interesting to me what's happening in this case.

the problem still exist after replace the DO_NOTHING_ON_CLOSE with DISPOSE_ON_CLOSE, if you use windows7 and it is ok, I can ignore this problem by use my windows10 computer. thanks.

closed 2019-02-17 by itviewer