hneemann/Digital #245: Brazilian Portuguese translation of static.xml

crcpucmg, 2019-02-13:
Attached translation of static_pt.xml for documentation purposes.

hneemann, 2019-02-14:
@crcpucmg 
Are you sure you have attached a file?

crcpucmg, 2019-02-15:
It's uploaded now: static_pt.zip.

Please, check.

hneemann, 2019-02-16:
I had some problems using the attached file because the encoding specified in the header does not match the encoding used in the file itself. I have trans coded the file from ISO 8859-1 (which I think is the default encoding used on Portuguese windows machines) to UTF-8 and hopefully it worked. To me it looks fine.
Perhaps you can also have a look at the attached PDF file: [Doc_Portuguese.pdf.zip](https://github.com/hneemann/Digital/files/2871763/Doc_Portuguese.pdf.zip)

commit 19edca44 (2019-03-13): updated portugues translation; closes #245; closes #260
closed 2019-03-13 by hneemann