hneemann/Digital #246: why the build in logic gate think of high z as 0

itviewer, 2019-02-17:
for example the Not gate,
[image]

The Not gate build with pmos and nmos transistors I learned from the textbook,
[image]
all built-in logic gates pull high z down to ground?  In a real circuit, is this the case?

hneemann, 2019-02-17:
No! The behavior of a real circuit depends on the technology used to implement the gates.
In your example of a typical CMOS implementation, a high Z input is dangerous and should be avoided. Due to the high input impedance of the FET gates, the input may drift to an intermediate voltage, making both the n-channel and p-channel transistors conductive. This can lead to a high current flow and damage the circuit. (See also [here](https://github.com/hneemann/Digital/issues/36#issuecomment-315969098))
In old chips which use TTL technology (multi-emitter bipolar transistors) a open input is interpreted as connected to high. I think that's why you often find a pullup resistor at CMOS inputs. The input then behaves like an old TTL input. But also a large pullup resistor increases the power consumption.

In Digital, all gates that are not specifically designed to deal with a high-z input read a high-z input as connected to zero.  But this has more programming reasons, because it makes some things easier to implement. Simply spoken: You can't draw current from a high-z input, so a LED stays dark. As if it were connected to zero.

itviewer, 2019-02-17:
thanks.
I now suspect that if we are not familiar with the physical characteristics of the Transistor devices, can we really design the digital circuit.

closed 2019-02-17 by itviewer

hneemann, 2019-02-17:
I think if a real chip is planned to have unconnected inputs, it's a good idea to consult the datasheet to check whether that's allowed or not.