hneemann/Digital #250: ATF150x-Latch

toms001, 2019-02-19:
Is it possible to build a latch in Digital for the ATF150x, analyze it and get it through the fitter?
Regards,
toms01

hneemann, 2019-02-19:
Every circuit which Digital is able to analyze, can be exported to a ATF150x chip.
If the circuit is large, it may not fit into a small device such as an AFT1502.
But in this case the fitter will display a appropriate message.

toms001, 2019-02-19:
i tried to build a latch like this:
https://www.semiversus.com/dic/grundlagen_der_digitaltechnik/latch_mux.svg
or like this one:
https://i.stack.imgur.com/T0gex.png
=> cycle-error.

Is there any possibility that should work in a ATF150x out of Digital?
Regards,
toms01

hneemann, 2019-02-19:
Both of your solutions are not ideal because you don't utilize the built-in flip-flops. And it's not a good idea to build a flip-flop your self. A better way is to build it that way:

[image]

The problem is that you need to use the clock pin like a normal logic input to control the mux. I have looked into the datasheet. The good news are: This is supported by the ATF150x series. The bad news are: This is prohibited by Digital.
In the next days I will try to see if it works if I allow it in Digital.

hneemann, 2019-02-19:
By the way: Do you just want to showcase a latch or do you plan to use it as a part of a more complex circuit?

toms001, 2019-02-19:
i'm trying to rebuild a replacement for a Commodore 6529 with a ATF1502 like this:
http://forum.6502.org/download/file.php?id=1909&sid=5256504fd121657f104b94f34196e99e

hneemann, 2019-02-19:
But as far i can see 74htc377 is not a latch. Its a octal flip flop with enable.

hneemann, 2019-02-19:
I looked at the data sheet of the Commodore 6529. The C6529 is a bidirectional port driver with flip-flop buffers. I think it can be replaced by an ATF1502, but you have to use the three state capabilities of the outputs, which are not supported by Digital. 
But it's possible to write a CUPL file to describe the logic and to use WinCUPL to create the JEDEC file required to program the chip.

hneemann, 2019-02-20:
> I have to deal with CUPL more closely to get a result.

Draw a simple circuit in Digital and create a CUPL file from it as a starting point. This makes it a lot easier.

> We tried the Keyboard-position with a 74xx573, but it didn't work.

Maybe i'm wrong, but if you want to replace the Keyboard-6529 which is used as input only, then a latch is not required. Also a ATF1502 is far to complex and expensive. A simple octal line driver with three state output should be sufficient to do the job.

toms001, 2019-02-20:
I got several ATFs as a gift and i think it is a good starting point for me to work with CPLDs.
It is a private project - not a commercial.

hneemann, 2019-02-20:
If the 74xx639 did not work, it seems very important to me to find out why not. 
Otherwise, I would expect the ATF1502 not to work either.

> I got several ATFs as a gift and i think it is a good starting point for me to work with CPLDs.
It is a private project - not a commercial.

The ATF1502 is a interesting chip and its fun to play with. But CUPL is not. Its an old first generation HDL similar to ABEL which is not used anymore. And WinCUPL is an old program, which also shows its age. I'm happy that I could remove CUPL/WinCUPL from the tool chain used in my lab exercises. Maybe I should find a way to support the three state capabilities of the ATF150x series.

toms001, 2019-02-20:
Yes, that would be nice. Unfortunately Prochip Designer needs a very expensive license for Mentors Precision Synthesis Tool.

I think i need latches and at the weekend i will test it with a logic analyzer.

hneemann, 2019-02-20:
Is there more comprehensive information about the MOS6529? The four-page data sheet is very brief.

toms001, 2019-02-20:
Not to my knowledge :(

hneemann, 2019-02-21:
I have played around with enable signal support. You can try version v0.21-98-g90c3214 which you can find [here](https://infdigital.dhbw-mosbach.de/).
This version allows to analyze a circuit like this:

[image]

If you create a JEDEC file the pin 11 controls the output driver of pin 24 and 25. The code (branch outputEnable) has some rough edges so I can't merge it. But i have tested it on a AFT1504plcc and it seems to work.

toms001, 2019-02-22:
Thank you very much, but i think we need latches (.LE) too. 
[image]
[image]
Do you want a logic analyzer file export? Which format? I'm using Pulseview.

hneemann, 2019-02-22:
From what I can see, a latch may not be necessary. But it doesn't matter: there is no CPLD out there that has a latch built in. The ATF1502 doesn't have that either. If necessary you have to build a latch, consisting of a flipflop and a muxer as shown above.

Can you take a higher resolution logic analyzer snapshot triggered at the falling edge of CS?

hneemann, 2019-02-22:
I have checked the datasheet of the ATF1502 and I was wrong: There is a latch mode available in that chip! I was not aware of that! And I'm surprised! ;-)

toms001, 2019-02-22:
Yes, thats what i mentioned. .LE
http://ww1.microchip.com/downloads/en/DeviceDoc/refdsgns.zip

[image]
[image]

I think it latched...

hneemann, 2019-02-22:
Hmm! To me it looks like triggered on the rising edge of CS. P0 does not change while CS is low. It changes on the rising edge of CS.  A latch would be transparent while CS is low, which means P0 should follow D0 as soon as CS goes low.
But the measurement resolution is not high enough to be sure.

toms001, 2019-02-22:
> And by the way: CS and RW are both low: The chip is used as an output. Not as an input as you mentioned above.

Yes, you are right.

[image]

Just found the sampling-rate of the LA.... sorry ... 12MHz for now

hneemann, 2019-02-22:
To me it looks like a flipflop/register triggered on the rising edge of CS.
And this observation also matches the data sheet.

toms001, 2019-02-22:
Thank you, you are right, i tested it with a 74LS273 (CS @ CP) - it works.
Now i have to find a way for the second position at the RS232/User-Port (2way).

hneemann, 2019-02-22:
I think it is possible to replace a 6529 with a ATF1502. The ATF1502 offers bidirectional pins, open collector outputs, and three state outputs.  That's all you need. Only if the 6529 contains pull up resistors you have to add those as additional components.

toms001, 2019-02-23:
For this, does Digital have to allow every signal (e.g. CS) at the flip-flops clock-pin?

hneemann, 2019-02-23:
Be careful when replacing the 6529 with a 74LS273! The 6529 has open collector outputs, whereas the 74LS273 has usual digital outputs which can drive current if set to high. The 6529 datasheet is not very clear but it seems to me the 6529 also contains pull up resistors. By replacing the 6529 with a 74LS273 you have replaced the pull up resistor with a short circuit.

I don't know how the keyboard logic works in detail, but if a other component (a keyboard key or a other driver) pulls the line usually driven by the 6529 down, it will lead to high currents. I think its worth to double check that.

> For this, does Digital have to allow every signal (e.g. CS) at the flip-flops clock-pin?

Digital is not a synthesis tool for the ATF150X series. The simulator uses the ATF150x chips to illustrate some logic during a lab exercise. That's it! Digital will never support all the features offered by the ATF150x series.  So I think: No, it won't.

hneemann, 2019-02-23:
Here is an example which shows the pull-up worried I have: _keyboard1.dig_ can cause a short circuit if two keys in the same column are pressed. If a open collector driver is used (_keyboard2.dig_) that does not happen. [keyboard.zip](https://github.com/hneemann/Digital/files/2896685/keyboard.zip)

toms001, 2019-02-23:
ok, thank you - i will investigate further.

btw.: A complete synthesis tools for the ATFs based at Digital would be very very good. I think Microchip should think about it to invest in project like this!

hneemann, 2019-02-24:
The ATF series is very old. Introduced by Philips in the mid '90s (called XC7000?) then Xilinx purchased the Philips CPLD division. Later Xilinx stopped the production, and Atmel jumped in. Then Atmel was purchased by Microchip. This is why the tools are so old and unmaintained. I don't think someone is going to much trouble to improve the support for these old chips.
All modern CPLDs come with modern design tools, free of charge for private use. But they all come in TQFP housing and most of them don't support 5V supply voltage, which makes them less suitable for a lab exercise.

closed 2019-02-24 by hneemann

toms001, 2019-02-24:
> I have played around with enable signal support. You can try version v0.21-98-g90c3214 which you can find [here](https://infdigital.dhbw-mosbach.de/).
> This version allows to analyze a circuit like [...]

Sorry again, i tested the version above with this circuit and exported it to cupl:
[image]

i got:
Device   f1502ispt44 ;

ar = 'b'0 ;

/* inputs */

/* outputs */

hneemann, 2019-02-24:
I have not implemented the CUPL export. Only TT2 export is implemented.
If you want to want to create a CUPL file, you can not use the OE function.
Create the OE signal as a normal output, then create the CUPL file and modify it manually.

Then it looks like this:
```
Name     z.dig ;
PartNo   00 ;
Date     21.02.2019 ;
Revision 01 ;
Designer helmut.neemann ;
Company  unknown ;
Assembly None ;
Location unknown ;
Device   f1502ispplcc44 ;

ar = 'b'0 ;

/* inputs */
PIN 43 = CLK;
PIN  9 = A4;
PIN 11 = A5;
PIN 12 = A6;
PIN 14 = A7;

/* outputs */
PIN 25 = B7n;
PIN 24 = B6;

enable = A5 & A6;

/* sequential logic */
B7n.D = A7;
B7n.ck = CLK ;
B7n.ar = ar ;
B7n.oe = enable;

B6 = A4;
B6.oe = enable;
```

One imported 'feature' you have to keep in mind: If you start the fitter, and the fitter can not fit the circuit to the selected chip, then the fitter performs an additional fitter pass where the fitter tries to fit the circuit by changing the pin numbers. If this is successful, WinCUPL does not inform you, that the pin numbers have changed! You end up with a JEDEC file which uses other pin numbers as defined in the CUPL file. Very frustrating if you are not aware of that. (Learned the hard way :-)
It is therefore a good idea to check the created pin file before testing the JEDEC file.