hneemann/Digital #251: How to simulate such an SRAM

itviewer, 2019-02-21:
Here is a real 32-bit asynchronous sram chip's [datasheet](https://www.cypress.com/file/46681/download),  there is a block diagram in the second page, and Truth Table in page of 15.
I still hope there is a 32-bit but support for partial writes (regardless of unaligned access) sram.

hneemann, 2019-02-21:
I don't understand the question! 
The chips you mentioned are not able to perform an unaligned access if a 32 bit border is crossed. But thats not surprising. This is exactly what you would expect from real chips. 
If you need such an access, model it in Digital, as described in #233. If you export it to an FPGA, it will work as expected.

itviewer, 2019-02-21:
I just want to know how a 32-bit SRAM that supports partial write function is constructed in a real chip. 
I think it should not be composed of four 8-bit SRAMs in a real chip.

hneemann, 2019-02-21:
> I think it should not be composed of four 8-bit SRAMs in a real chip.

Logical it does. There are no separate chips of course. All the SRAM is created on the same die. But there are four banks (or how ever you would call it) which have separate enable logic. They only share the same address lines.

Imagine this all being on a single die:

[image]

If you are interest in how a simple sram cell works you can look at "examples/cmos/sram.dig".

itviewer, 2019-02-22:
If there is an array `char str[] = {'a','b','c','d'}`;  suppose the starting address of the array is 0x0, how does the above memory structure store the array?
[image]
or
[image]

hneemann, 2019-02-22:
Look at the datasheet you mentioned earlier. The title is "16-Mbit (512 K words × 32 bits) Static RAM".
According to the datasheet the chip has 19 address lines (A0-A18). This means the chip can deal with 2^19=512\*1024 (512k) different addresses. To store 16-Mbit, you have to address 32 bits with each address value because 512\*1024\*32 gives 16Mbits. 

Together this means the memory layout looks like this:
[image]
Every address value accesses 32 bit. The same address layout is generated by the Digital circuit shown above.

If you want to use such a memory layout:
[image]
you need a chip called "16-Mbit (2 M words × 8 bits) Static RAM". Such a chip requires 21 address lines (A0-A20). Then its possible to address individual bytes. But you can't read/write four bytes at once.
Or, if you want to use a chip like a "16-Mbit (512 K words × 32 bits) Static RAM", you have to add some fancy control logic as shown in #233 to make byte addressing possible.

closed 2019-02-22 by itviewer