hneemann/Digital #252: 28c256 EEPROM - Write enable inverted

IvyBit, 2019-02-24:
Thank you for the work you have put in this project!

I'm probably just not getting it but the 28c256 (or any other eeprom) only seems to store data when the
/WE pin goes from low to high. 
I took a look at the internal eeprom and it seems to be setup for a high to low pulse, but I can only get it to actually save data with an inverter between the clock and /WE.

hneemann, 2019-02-24:
I think you are right!
A real 28c256 latches the address on a falling edge of /WE and the data is latched on the rising edge of /WE.
But the simulation latches address and data both on the falling edge of /WE.

IvyBit, 2019-02-24:
Thank you for clarifying this! I thought I was going crazy.
Are you planning to update the eeprom lib?

hneemann, 2019-02-24:
Not sure yet. Maybe I have to fix this in the java code.

hneemann, 2019-02-25:
Please try the latest [pre release build](https://infdigital.dhbw-mosbach.de/).
Please use v0.21-97-g657164e.

closed 2019-02-25 by hneemann

IvyBit, 2019-02-25:
Sorry couldn't test it earlier.
Thanks again for this awesome project!

commit 657164e4 (2019-02-25): fixed a timing issue in the EEPROM, closes #252