hneemann/Digital #253: HDLNet only allows one output but multiple inputs

L3tum, 2019-02-24:
First of all, I love your program. I'd been searching ages for something like this and it's literally a godsend. I had signed up for a class called "Physik Technik" where we did a lot of microeletronic stuff but always bought the parts, which made the whole thing quite tedious. 

Anyways, I recently tried to export an instruction decoder to VHDL and got greeted by the error "multiple outputs connected to net null". When I search in the source code for that only [this](https://github.com/hneemann/Digital/blob/master/src/main/java/de/neemann/digital/hdl/model2/HDLNet.java#L50) class shows up, which is the HDLNet class. According to your class comment, it's supposed to be able to have multiple outputs, but only one input. However, in the `addPort` method it throws an error when another *output* is added.

I could be totally wrong about this though, as I have no idea about the structure of your program.

hneemann, 2019-02-24:
Could you please attach the circuit which does not work? 
Then I can figure out what's going on.

L3tum, 2019-02-24:
It's actually an issue on my end. I was under the impression that any output that isn't used needs to be connected to ground and that's what was creating the errors.

Sorry, and thank you!

closed 2019-02-24 by L3tum