hneemann/Digital #255: Error in class PlainSwitchDT?

DmitryVyal, 2019-02-26:
When modeling the individual components of early electronic calculators, a strange circuit modeling error was detected. After debugging, the minimal fragment of the scheme in which the error is reproduced is highlighted.
[image]
The circuit on the left is working correctly, and the circuit on the right is modeled with an error. After switching the state of the model is clearly not as expected.

hneemann, 2019-02-26:
There is an optimization which improves performance if constant signals are switched.
In special cases this optimization did not work, so the circuit with the constant did not work properly either.
Could you please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) build.
I'm not sure if i have fixed all the issues. 
Please try carefully, and please let me know how it works out.

DmitryVyal, 2019-02-26:
An error occurs when one of the pinB and pinC contacts is connected to a constant value, and the second with a non-constant value. Adding the check of the third contact of the switch to the PlainSwitch.createSwitchModel function allows to solve this problem. This change disables optimization for some schemes that have problems.
[PlainSwitch.diff.txt](https://github.com/hneemann/Digital/files/2906649/PlainSwitch.diff.txt)
[PlainSwitchDT.diff.txt](https://github.com/hneemann/Digital/files/2906650/PlainSwitchDT.diff.txt)
Perhaps the condition is not entirely correct as it does not take into account the case when input3 == null. This can lead to excessive optimization constraints.

hneemann, 2019-02-26:
Could you attach a circuit which does not work with the latest pre release version?

commit ec1fce6f (2019-02-28): fixes #255
closed 2019-02-28 by hneemann