hneemann/Digital #258: Test data "To many entries!"

deyspandan, 2019-03-03:
Is there some rational behind keeping the "MAX_RESULTS" in TestExecutor class to 1024 (1 << 10)

I have some really long test vectors, hence i rebuild changing MAX_RESULTS to 32768 (1 << 15)
The application build successfully and apparently the test simulations were successful!!

Forgive me this might be a noob question, I am not so familiar with Java.

Great application thanks a ton!!

hneemann, 2019-03-04:
It makes no sense to display so many results in the table. No one could ever inspect all these entries.
But as long as no error is displayed in a separate error dialog, all test cases will be executed, and if one fails, these rows will still be displayed.

deyspandan, 2019-03-04:
You are right. It is difficult to inspect them manually hence I create the test vectors using a python script.

My test vector has the following structure: 
**Write(n bytes)->clock(more than 1024 times)->Read(n bytes)**

The majority of the cases are clock ticks.
So far this has worked fine and no errors have popped up.

hneemann, 2019-03-04:
Please read the help text which is available in the test case editor.
Its easy to create loops in the test case description to do just that.

deyspandan, 2019-03-04:
I do use the given "repeat(n)" primitive for my clocking
But as I understand this will generate n test lines and this is where it truncates the test vector with the comment "To many entries!"

Basically I am designing an interleaver circuit and way I test is as follows:

1. I load the RAM with n Bytes of data (as 32 bit words)
2. Clock the circuit the number of bits in the data (n Bytes * 8) + some setup (~5 Clocks)
3. Read data out and match (as 32 bit words)

If my Step 2 gets larger than certain steps I cannot go to Step 3 and verify.
Hence the requirement for a large table :-P

hneemann, 2019-03-04:
> But as I understand this will generate n test lines and this is where it truncates the test vector with the comment "To many entries!"

The test vector is never truncated, the test vector is always executed completely. Only the results are not displayed completely if the expected values match the actual values.

But in a test case you always specify the expected values. And if the expected values will match, it is not necessary to check the values manually. And if they don't match, the results will be displayed anyway.
Maybe I haven't really understood the use case yet.

deyspandan, 2019-03-04:
App details
Digital.jar
Build git-Revision: v0.21
Build Time: 2018-12-10 09:47

> But in a test case you always specify the expected values. And if the expected values will match, it is not necessary to check the values manually. And if they don't match, the results will be displayed anyway.

Here is a snapshot of the test results
As you can see the test for k_1504 it failed but it only shows till line 340 and does not show the failed cases, which should be towards the end (line ~1550 -1596)
[image]

But when I rebuild the app with the "MAX_RESULTS  1 << 15" modification
You can see the failed test cases

App details
Digital.jar
Build git-Revision: v0.21-111-g8352fd2-dirty
Build Time: 2019-03-03 22:38

[image]

hneemann, 2019-03-04:
This is a bug! Can you please attach the circuit to the ticket?

hneemann, 2019-03-04:
No example circuit required! I was able to figure out for myself what was causing the problem.

closed 2019-03-04 by hneemann

deyspandan, 2019-03-04:
Thanks for looking into it!

Cheers!!

commit c3c29b82 (2019-03-04): fixes a bug in limiting the display of test results, closes #258