hneemann/Digital #263: Import external module

itviewer, 2019-04-10:
It is a bit of a hassle to manually write the code inside the module. 
It will be very convenient allowing reading the content from external files(vhdl or verilog), So we don't need to copy the content to Digital every time the file is modified. 
In addition, it is more better if we can automatically read the port definition (input, output) in the file.

hneemann, 2019-04-10:
What about right clicking on the editor? This allows you to load a file into the editor.
And if you press the check button at the bottom the input/output section is filled automatically.

itviewer, 2019-04-10:
ok,  it seems that auto fill input/output for verilog doesn't work
```
module test(a,b,y);
  input a,b;
  output y;
  assign y = a & b;
endmodule
```
[image]

hneemann, 2019-04-10:
Up to now the integrated verilog parser does not support this notation.
Try this:
```
module test (
  input a,
  input b,
  output y );
  assign y = a & b;
endmodule
```

commit b080837e (2019-04-13): improves the verilog parser, closes #263
closed 2019-04-13 by hneemann