hneemann/Digital #266: Circuit statistics

kendfrey, 2019-04-19:
I would like to see the number of each type of component in my circuit. For example, a circuit with 3 NAND gates and 2 copies of an embedded circuit containing 4 NAND gates would display as having 11 NAND gates.

This would be a useful stat for people who want to optimize circuits for the minimal number of gates, transistors, etc.

See also "Get Circuit Statistics" in: http://www.cburch.com/logisim/docs/2.5.0/en/guide/menu/project.html

hneemann, 2019-04-19:
Hmm! To implement a CMOS NAND gate, for example, [inputs]*[bits]*2 transistors are required if the number of inputs is small. Otherwise, multiple gates must be used. 
To calculate how many transistors are needed, at least the number of bits and the number of inputs are required.

hneemann, 2019-04-20:
I have added a simple statistics function. It works a bit different than you suggested:
The number of transistors needed is calculated. Since this may depend on the implementation technique, it is possible to set a transistor count in the circuit settings.
If a subcircuit has set a transistor count, this value will be applied. If not, the transistor number is calculated from the components used in this circuit.
Try the latest [pre release built](https://infdigital.dhbw-mosbach.de/).

commit 59302512 (2019-04-21): updated the release notes, closes #266
closed 2019-04-21 by hneemann