hneemann/Digital #272: External VHDL identifiers are formatted by LaTeXFormatter

smay4finger, 2019-04-25:
When I use identifiers with an Underscore, the LaTeXFormatter is making a subscript out of this Underscore. This looks very weird. I could try to fix it, when it's not a commonly used feature. What's the opinion about this?

hneemann, 2019-04-26:
Please explain your use case in a bit more detail.

smay4finger, 2019-04-26:
The following VHDL code:

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

entity NAME_WITH_UNDERSCORE is
    port(
	signalA_with_underscore: in std_logic;
	signalA: in std_logic;
        Q: out std_logic
    );
end NAME_WITH_UNDERSCORE;

architecture BEHAVIOUR of NAME_WITH_UNDERSCORE is
    constant PERIOD: time := 100 ms;
begin
    process begin
       
    end process;
end BEHAVIOUR;
```
looks like this:
[image]

but should more look like this:
[image]

I made the following change to achieve the above result:

```patch
diff --git a/src/main/java/de/neemann/digital/draw/graphics/text/Parser.java b/src/main/java/de/neemann/digital/draw/graphics/text/Parser.java
index f1c16411..57501792 100644
--- a/src/main/java/de/neemann/digital/draw/graphics/text/Parser.java
+++ b/src/main/java/de/neemann/digital/draw/graphics/text/Parser.java
@@ -90,7 +90,7 @@ public class Parser {
                     getChar();
                     sentence.add(new Decorate(parse('$'), Decorate.Style.MATH));
                     break;
-                case '_':
+                case '%':
                     getChar();
                     sentence.getIndex().addSub(parseBrace());
                     break;
```

hneemann, 2019-04-26:
Such a modification is not a good idea. I like the formatting of the text in LaTeX style. It's intuitive and well known.
The introduction of a '%' for entering a subscript will also break many existing circuits.
Maybe it is better to disable formatting only in the external component.

smay4finger, 2019-04-26:
Yeah, that was just a workaround. Didn't find a better way to disable subscript for showing the expected behaviour. I also like the formatting, but for the identifiers it's not a good idea.

Am 26. April 2019 20:16:51 MESZ schrieb Helmut Neemann <notifications@github.com>:
>Such a modification is not a good idea. I like the formatting of the
>text in LaTeX style. It's intuitive and well known.
>The introduction of a '%' for entering a subscript will also break many
>existing circuits.
>Maybe it is better to disable formatting only in the external
>component.

hneemann, 2019-04-26:
Please try the latest [pre release built](https://infdigital.dhbw-mosbach.de/).

smay4finger, 2019-04-26:
Yep, works exactly as expected. Thank you very much.

closed 2019-04-27 by hneemann

ProxyPlayerHD, 2019-07-13:
I'm still having the same problem, even with the newest pre-release.
overall using an underscore was a really bad choice seeing as it's very often used to name things in circuits/programs. why not use something that is rarely used?

hneemann, 2019-07-13:
@ProxyPlayerHD 
I like the LaTeX formatting of expressions. It's very handy when it's possible to name a pin like "Q_i^{n+1}".
And on the other hand, as in other programming languages, I use camelCase to name files or identifiers. That's why the problem doesn't concern me so much.

ProxyPlayerHD, 2019-07-13:
but what about a lot of other people who don't want to use it and just want to name their stuff?
why not add an option to disable the weird formatting?

hneemann, 2019-07-13:
> why you would even need to format names is beyond me to begin with...

Because "~Q" and "D_0" look much better when properly formatted!  :smile: And that is important for me when I create teaching material.

> so why not simply add an option to disable the formatting? that way everyone wins

Unfortunately, for a proper implementation, all text output code would have to be refactored to keep Swing, SVG, SVG/LaTeX, and sizing consistent. And of its very nature, the test coverage of the formatting code is low, making changes even more difficult. This seems too much effort just to allow an underscore in signal names. But I'll think about whether there's an elegant, simple way.

hneemann, 2019-07-14:
@ProxyPlayerHD 
What about adding an option to inputs and outputs to disable the formatting of the label of that specific input or output?

ProxyPlayerHD, 2019-07-15:
>Because "~Q" and "D_0" look much better when properly formatted!
i can understand the ~Q but why the D with a smaller 0 next to it? that just seems strange as i've never seen anyone format names like that.... especially when they are not numbers... like "DATA_IN_CPU"

>What about adding an option to inputs and outputs to disable the formatting of the label of that specific input or output?
yea that'd be fine by me. will it also work for components? like Registers?

also i got a few suggestions, can i just put them as a list (with explaination to each) in a single issue? (a new one, not this one)

hneemann, 2019-07-15:
@ProxyPlayerHD 

> i can understand the ~Q but why the D with a smaller 0 next to it? that just seems strange as i've never seen anyone format names like that.... especially when they are not numbers... like "DATA_IN_CPU"

Digital is primarily intended for use in education, which - at least for me - increases the importance of mathematical notations.
This means that something like this should be possible:
[image]

> yea that'd be fine by me. will it also work for components? like Registers?

Are you planning to export your design to VHDL or Verilog? If the first character of a label is a backslash, the formatting is disabled. But up to now the backslash is also exported to a HDL.

> also i got a few suggestions, can i just put them as a list (with explaination to each) in a single issue? (a new one, not this one)

I always appreciate suggestions. You can submit them as you like.

ProxyPlayerHD, 2019-07-16:
>Digital is primarily intended for use in education, which - at least for me - increases the importance of mathematical notations.
This means that something like this should be possible:

oh i see

>Are you planning to export your design to VHDL or Verilog? If the first character of a label is a backslash, the formatting is disabled. But up to now the backslash is also exported to a HDL.

i did not know that was a feature, and yes i was planning of exporting to Verilog, however that works

>I always appreciate suggestions. You can submit them as you like.

sweet, gonna write something up and post it whenver i have time

commit 1c24953a (2019-04-27): disables text formatting in external component, closes #272
Now a backslash in front of a text disables the string formatter.