hneemann/Digital #274: Overlapping input and output should behave as connected

smay4finger, 2019-05-02:
When I draw an input and output overlapping, they are not connected. You explicitly need to draw a wire from input to output, which is sometimes inconvenient. Is there a possibility to implement this behaviour? In the attached screenshot the lower circuit is broken, because not connected.

[image]

hneemann, 2019-05-03:
I looked at it: This enhancement is surprisingly complicated to implement and adds a lot of complexity. 
I don't think it's worth the effort.

hneemann, 2019-05-04:
I have added a simple auto wire function which creates the necessary wire automatically if a new component is inserted. Maybe this is also helpful.
Please try the latest [pre release version](https://infdigital.dhbw-mosbach.de/).

smay4finger, 2019-05-07:
Hmm, doesn't work as expected. Tried version v0.22-59-g7bb0b24.

smay4finger, 2019-05-07:
And something worse happened. The auto-wire functionality is breaking components with nearby pins beside each other. I inserted e.g. a seven segment LED and the two inputs got somehow connected to each other.

hneemann, 2019-05-07:
I have fixed the "seven segment bug". Please try v0.22-64-g67bdce0.
But a wire is still required. But this wire is created automatically when two pins are spaced from each other by one grid width.

[image]

If you place two components like this, the wire is created automatically.

smay4finger, 2019-05-08:
Ok, this auto-wire works. Still it's not possible without a wire, but if that's to complicated, so be it.

commit f3f6c7d0 (2019-05-09): Adds a simple auto wire function; closes #274
closed 2019-05-09 by hneemann