hneemann/Digital #275: Digital is not saving settings

YulienPohl, 2019-05-08:
Hi, every time when I reopen Digital the checkboxes in the properties window are reseted 
but not the other settings like language or the library paths.
Furthermore it be would good if the window size is saved after closing.

hneemann, 2019-05-08:
All other settings are saved, but the checkboxes in the same dialog are not?

YulienPohl, 2019-05-08:
Yes, for example Component tree view.
This problem occurs also on my other windows machine.

hneemann, 2019-05-08:
In your home folder is a file ".digital.cfg" stored.
Does this file change if you modify the check boxes?
Do the made changes affect this file?
The file should be saved if the settings dialog is closed.

YulienPohl, 2019-05-08:
I have found the problem.
I had an older .digital.cfg file version in the folder which I deleted, now it saves the settings.
But the windows size is not saved as fullscreen.

commit 4b60be08 (2019-05-09): Window size is stored and restored at startup; closes #275
closed 2019-05-09 by hneemann