hneemann/Digital #276: native file dialog when opening a file

jones1008, 2019-05-15:
When opening a file via `file - open` it shows an open file dialog of the Java JFileChooser API.
It would be much more user friendly if the native file dialog is shown based on the operating system because thats the dialog the user is used to.
Here it is described for windows on this [stackoverflow question](https://stackoverflow.com/questions/39819319/windows-native-file-chooser-in-java).

3TUSK, 2019-05-15:
Had a quick look at the stackoverflow question. The answer is using JavaFX, but IIRC Digital only uses Swing. Further, that answer also mentioned that such solution will not work when mixing JavaFX and Swing, so I doubt how feasible is to address this issue.

hneemann, 2019-05-15:
@3TUSK You are right: Digital is based on Swing. And Swing is not designed to open native UI components. 
It would be very complex to implement. I don't think the benefit justifies the effort.

closed 2019-05-17 by hneemann