hneemann/Digital #277: can't change true table output Y name

foxzzu, 2019-05-20:
[image]
with mouse right click the output Y, no respond. The system is macOS 10.14.5

hneemann, 2019-05-20:
Please try the pre-release built v0.22-75-g23ff923 which you can find [here](https://infdigital.dhbw-mosbach.de/).
It should be able to detect the MAC secondary click (CTRL+Left).

foxzzu, 2019-05-20:
pre-release built v0.22-75-g23ff923, the output name can change. And, when mouse click the menu, there is no responding.
[image]

hneemann, 2019-05-21:
Your screenshot looks really weird!
I suspect that there are files in your home folder with names similar to "Digital_2019-05-12_15-42-04.log".
If there are such files, can you please zip them and attach them to this ticket?

hneemann, 2019-05-23:
@foxzzu 
Please try the latest pre release build v0.22-113-g25c3001. 
I've fixed some issues that could be the cause of your problems.

hneemann, 2019-05-25:
@foxzzu 
Do not hesitate to reopen the issue if new problems arise.

closed 2019-05-25 by hneemann