hneemann/Digital #281: MacOS scrollbars?

dmolony, 2019-06-18:
When a circuit becomes larger than the screen, should scrollbars appear? I cannot work out how to move the canvas around, all it seems to do is zoom in and out.

hneemann, 2019-06-18:
The secondary mouse click should allow to move the canvas.
It is also possible to zoom out until the interesting section of the circuit is visible, then point to the interesting region and zoom in again.

dmolony, 2019-06-18:
The secondary mouse click just wants to draw rectangles. The click-zoom seems to work, but it's a little clumsy and always stays in line-drawing mode afterwards. Is there any reason behind not having a scrollable canvas?

hneemann, 2019-06-18:
If you want to add scrollbars you need a size of the canvas.
On Windows a right press allows moving the canvas even if a component is selected. In the settings you can enable windows mouse handling.

dmolony, 2019-06-18:
I would have thought it would be harder to not know the size of the canvas.

I would like to request a new feature - scrollbars!

Every document has a fixed size expressed in a height and width in page units. The user chooses the page size (A4, US letter etc), and the document starts out as 1 x 1 page units. As the circuit grows or shrinks, so does the canvas. So it might grow to 2 x 1 pages, or 4 x 2 etc.

And now the canvas size in pixels is known, which presumably would be multiplied by the zoom ratio.

hneemann, 2019-06-20:
Your suggestion seems very complicated to me if you consider that it is only about the scrollbar display.

hneemann, 2019-06-22:
@dmolony 
Please try the latest pre release built you can find [here](https://infdigital.dhbw-mosbach.de/).

dmolony, 2019-06-22:
Love it!

commit 016c7afa (2019-06-22): updated release notes, closes #281
closed 2019-06-22 by hneemann