hneemann/Digital #282: How do I get the clock to auto-run?

keller91711, 2019-07-13:
I don't see any documentation for how to start the clock auto-running. I only seem to be able to toggle it single-step.

hneemann, 2019-07-13:
The clock component has a check box "Start real-time clock".

keller91711, 2019-07-13:
I'm sorry, but I don't see any check box.

keller91711, 2019-07-13:
Never mind, I found it, with control-click. It is not obvious that I should have. The pdf documentation doesn't mention control-click.

closed 2019-07-13 by keller91711

hneemann, 2019-07-13:
You're right! The documentation says "right click", but on MacOS "control click" is used. Perhaps this detail should be added.

keller91711, 2019-07-16:
Thanks.
Are you open to discussing the interaction design?
Bob Keller

hneemann, 2019-07-16:
I'm always open to suggestions. This is the only way the simulator can evolve.
But to be honest, the simulation core itself is the most interesting part for me. I find the GUI rather a necessary evil.
But of course I know that the GUI is the part that users perceive more.

keller91711, 2019-07-21:
So far, for me, the most irritating part is having it draw lines without
any key being held. This results in lots of erasures of unintentional
lines. But there are other things.

Bob

hneemann, 2019-07-21:
You're right. Maybe I should change that. In my imagination I click where the wire has to be soldered and with the right mouse button I cut the wire. So the UI feels quite natural.
But most people will rather think of drawing a line with a pen and try it as they are used to from other drawing programs.

keller91711, 2019-08-05:
This may be a bug, I'm not sure. If it isn't, I don't understand.
I'm synthesizing a 3-input, 3-output combinational function. I enter the
truth table:
[image: image.png]
then press Create>Circuit, but I get a circuit with the A input ignored,
which is surely incorrect.
[image: image.png]
[image: image.png]
I also had trouble figuring out how to name the outputs other than Y. The
way I did it was to create a dummy circuit with the desired labels first
but this seems very cumbersome.

Thanks for any help.

Regards,

Bob Keller

keller91711, 2019-08-05:
Never mind, I think I see it. By accident, the table is degenerate in input
A, so it doesn't get shown.

Bob

hneemann, 2019-08-06:
With a secondary click on the table header you are able to edit the input and output names.

keller91711, 2019-08-06:
On the Mac, control-click on the column headers doesn't seem to do anything.

Bob

hneemann, 2019-08-06:
There was a problem on MacOS which is fixed but not released yet. Please see #277 and try the latest pre release.

keller91711, 2019-08-06:
Thanks. I'm really enjoying using the system. I will be using it for the
logic part of my advanced intro CS class in September.

Regards,

Bob

hneemann, 2019-08-06:
> Thanks. I'm really enjoying using the system.

You are welcome.

> I will be using it for the logic part of my advanced intro CS class in September.

Please let me know how it worked out!

keller91711, 2019-08-07:
Helmut,

I think something is incorrect with the custom components file menu. When
it gets to a certain size, say more than 12 files, some files behave as if
they were directories, with other component files below them. This makes a
particular component harder to find.

Bob

hneemann, 2019-08-08:
See #152