hneemann/Digital #283: some suggestions

ProxyPlayerHD, 2019-07-17:
- Enable input for Multiplexers, De-multiplexers and Decoders

i mean really, why not? i have to do weird stuff with controlled buffers to get the same effect

- Reset input for Registers

again, why not? it seems like an important thing to have if you don't want the circuit to only reset via stopping the simulation

- A better hex editor for Memory and similar

it's quite inconvenient as you cannot even copy and/or paste stuff from/to memory easily or at all. good example is HxD

- ability to pre-fill data into RAM

basically so that you can load a file or copy/paste data into the RAM component, and it will always be set to those values at the start of the simulation. i mean you can do the same in verilog if you really want to, so why not here?

lastly:
not really a suggestion but rather a question, how do i debug circuits that have a lot of subcircuits? i cannot run the simulation and look into subcircuits at the same time like in logisim, so debugging is really difficult. i tried to recreate behaviors of inputs while inside a subcircuit but it never gets to the same result.
[Here is the circuit i'm trying this with](https://www.dropbox.com/s/cz7ac40evf4i43e/Digital%20CPU.zip), you can try it out if you want. it also includes the Logisim circuit i based it off.

 the exact problem i've having is that the ALU of this CPU is misbehaiving in unexcepted ways. i've tested it multiple times and it works correctly when i only test the ALU. but with the rest it doesn't seem to work, the "hello world" program inside the Logisim will also work with this CPU (obviously because the Digital one is based off the Logisim one) but it fails at the "INC (**)" Instruction where instead of incremementing a byte in Memory it seems to add it to A and store that in Memory. i have not tested more math Instructions because i know that if this one works all should work.

anyways, i hope the frustration at the end wasn't too long i just really want to somehow implement this CPU in verilog but i cannot fiquire out how so i tried it with Digital to export it and check how it's done there

hneemann, 2019-07-17:
> Enable input for Multiplexers, De-multiplexers and Decoders

As you mentioned: It is possible to add a controlled buffer to the output.
And a decoder with an enable input is essentially the same component as a demultiplexer.

> Reset input for Registers

You can use a multibit async D-FlipFlop which has an async clear. And a sync clear can be implemented with a register, a muxer and an OR gate.

> A better hex editor for Memory and similar

That's a good point! But I expect most users use external programs to modify/create the data in a file and simply load that file to a ROM.

> ability to pre-fill data into RAM

If you declare a RAM as programm memory, you can select a file in the circuit settings which is loaded to the RAM when the simulation is started.

I looked at your processor: There are some problems with it that can explain why the processor works in Logisim but not in Digital.
1. There are logic inputs which are driven by a gated buffer, and the buffer drives only this logic input, nothing else is connected to the input. Logisim and Digital behave different when logic inputs are set to high-z. Digital reads always a zero. Why not replacing such buffers with an AND gate?
In the CU the ROM address can be switched to high-z. Why is that?
2. There are some clocked components which are driven by gated clock signals. This can cause lots of hard to debug timing issues. Especially because the propagation times are different in Logisim and Digital.
3. Your reset pulse is very short. It is better to use the built-in reset generator. It adjusts its reset time to the circuit.

ProxyPlayerHD, 2019-07-18:
>As you mentioned: It is possible to add a controlled buffer to the output.
And a decoder with an enable input is essentially the same component as a demultiplexer.

but the point was that it would be much easier if the component had it built in aready, same with the register reset, i mean counters have them too...

>There are logic inputs which are driven by a gated buffer, and the buffer drives only this logic input, nothing else is connected to the input. Logisim and Digital behave different when logic inputs are set to high-z. Digital reads always a zero. Why not replacing such buffers with an AND gate?

buffers are more compacts and convenient than a bulky AND gate. and since High-Z is the same as a 0 it just makes that even more convenient to use than an AND gate.

>In the CU the ROM address can be switched to high-z. Why is that?

you mean in Logisim? honestly, not sure... in Digital it makes sense since both the ROM and the constant just go onto the same line.

>There are some clocked components which are driven by gated clock signals. This can cause lots of hard to debug timing issues. Especially because the propagation times are different in Logisim and Digital.

what/where do you mean? everything should run off the actual clock signal, except the ALU which is supposed to wait a bit after the rising edge before saving anything, and since registers cannot be set to update during a High level in Digital it's a lot harder to do.

>Your reset pulse is very short. It is better to use the built-in reset generator. It adjusts its reset time to the circuit.

i tried the built in Reset generator but it stayed high constantly and never turned off. where do you adjust it? when you look at the attributes it only has an option to invert the output, give it a label, and rotate it.

hneemann, 2019-07-18:
> buffers are more compacts and convenient than a bulky AND gate. and since High-Z is the same as a 0 it just makes that even more convenient to use than an AND gate.

See #36

> what/where do you mean? everything should run off the actual clock signal, except the ALU which is supposed to wait a bit after the rising edge before saving anything, and since registers cannot be set to update during a High level in Digital it's a lot harder to do.

It is dangerous if the clock signal is not directly connected to the clock inputs of all clocked components. See an example [here](https://github.com/hneemann/Digital/issues/97#issuecomment-355736446).

> i tried the built in Reset generator but it stayed high constantly and never turned off. where do you adjust it? when you look at the attributes it only has an option to invert the output, give it a label, and rotate it.

The reset output is set to low or high during the stabilisation phase of the circuit immediately after power-on. You will never see a value change in the simulation, since the first drawing update takes place after the reset pulse has elapsed.

ProxyPlayerHD, 2019-07-18:
>See #36

i mean i see why flaoting pins and inputs are bad, but why can't i just add a pull down resistor somewhere on the same wire? wouldn't it then work as an AND gate since the output is either 0 or whatever the input is

>It is dangerous if the clock signal is not directly connected to the clock inputs of all clocked components. See an example here.

alright so i changed some parts of the circuit, but not many actually have clock signals that go through a gate.

also since the actual clock component itself goes through and AND and a OR before it even reaches a single component i assume it's alright since there is no delay difference for anything?

>The reset output is set to low or high during the stabilisation phase of the circuit immediately after power-on. You will never see a value change in the simulation, since the first drawing update takes place after the reset pulse has elapsed.

ok the help page really confused me, because the component is by default inverted i thought the description still applies normally, but aparently the description applies to the uninverted Reset thing.

also implementing all changes you said... somehow made the CPU work correctly. what.
Thank you though, it feels good to have something finished.
now i can try a more complex CPU. oh boy.

lastly, a suggestion that just came to my mind while thinking of uses for my "faster than logisim" CPU, how about always correct scaling for the Graphic RAM window? you can currently stretch it however you want, which deforms the Pixels and makes it basically impossible to get to the correct aspect ratio again without reseting the entire simulation, so how about it does the same as the Windows Media player where it add black bars to either site to keep the correct aspect ratio?

hneemann, 2019-07-18:
> also since the actual clock component itself goes through and AND and a OR before it even reaches a single component i assume it's alright since there is no delay difference for anything?

That's right.

> also implementing all changes you said... somehow made the CPU work correctly. what.
> Thank you though, it feels good to have something finished.

You're welcome. :smile:

> lastly, a suggestion that just came to my mind while thinking of uses for my "faster than logisim" CPU, how about always correct scaling for the Graphic RAM window? you can currently stretch it [...]

Hmm! I'll think about it!

ProxyPlayerHD, 2019-07-18:
ok after exporting the CPU to Verilog i noticed that everything is inside a single file, is there an option to make every module have it's own file? it would make it much easier to read

i also noticed that the ROM is being turned into an Array with every single value in it. which is the easiest way to do it code wise, but how much harder would it be if it were to use a case statement where only the addresses that have data in them get a case, and everything else just falls to default and gets set to 0?

so for example you have this 8 bit ROM with 4b addresses:

`0....1....2....3....4....5....6....7....8....9....A....B....C....D....E....F`
`0x00 0x00 0x42 0x6E 0xFF 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x8A 0x00 0x00 0xFA`

so the case statement would looks like this:
```
case (address)
    'h2 : data <= 'h42
    'h3 : data <= 'h6E
    'h4 : data <= 'hFF
    'h9 : data <= 'h10
    'hC : data <= 'h8A
    'hF : data <= 'hFA
    default : data <= 'h00
endcase
```
which is much more compact and probably also takes up less space on an actual FPGA

hneemann, 2019-07-19:
> ok after exporting the CPU to Verilog i noticed that everything is inside a single file, is there an option to make every module have it's own file?

No, and also not planned.

> i also noticed that the ROM is being turned into an Array with every single value in it. which is the easiest way to do it code wise, but how much harder would it be if it were to use a case statement where only the addresses that have data in them get a case, and everything else just falls to default and gets set to 0?
> ...
> which is much more compact and probably also takes up less space on an actual FPGA

A ROM is more or less a simple locigal function which can be represented by LUTs. In your case most of the 27 output bits depend on all 13 input bits (only bit 7 is always zero, and bit 8 is an AND of all 13 input bits). I don't beleve that the kind of Verilog code which is used to represent the function does effect the FPGA logic cell consumption much. HDL analyzers are pretty good these days.

What would make a big difference would be the definition of a large number of "don't cares". This can reduce resource consumption a lot.

closed 2019-07-26 by hneemann

ProxyPlayerHD, 2019-07-27:
>No, and also not planned.

aww that's sad, i like having clean code... but it can be done manually just a little more effort.

>A ROM is more or less a simple logical function which can be represented by some LUTs. In your case most of the 27 output bits depend on all 13 input bits (only bit 7 is always zero, and bit 8 is an AND of all 13 input bits). I don't beleve that the kind of Verilog code which is used to represent the function does effect the FPGA logic cell consumption much. HDL analyzers are pretty good these days.

i see, so it only looks like a huge chunk inside the file