hneemann/Digital #284: Improved Single-Stepping

dfbg, 2019-07-18:
It would be really helpful if there was a button that would run the steps automatically to the next pause (when there are no more blue ovals).  When working with my clocked circuits there are many cycles that are idling waiting for a divided clock transition to arrive.

It would also really help if the single step and this new button had keyboard shortcuts.

And I find myself clicking the Run Tests accidentally way too frequently.  Probably because it has a play arrow like the other buttons it's right next to and has green on it like the running wires.  It's like it's magnetic.  The keyboard shortcuts would help with this.  As would disabling the button while in step mode.

hneemann, 2019-07-18:
Does this help? Maybe the break component should be used in single step mode somehow. Hmm!!

dfbg, 2019-07-18:
The circuit I'm working on right now has the main behavior triggering at clock/8.  Having to manually step through the 6 "idle" clocks cycles gets really tedious when it has to be done with the mouse.  I can't even use the clock keyboard shortcut because the focus moves to the toolbar when I click the button.  Those "idle" cycles are used by other parts of the circuit to process input without colliding with the /8 behavior and I still need to watch them occasionally. 

Some way to streamline toggling the clock and then auto-clicking the step button until it gets disabled.  Typing "c", "f" (finish), "c", "f", etc until it gets to the cycle I need to look closely at would save a lot of time.

Having Break work in single step would also be useful but not a general substitute.  I think there will be situations where the conditions are just too complicated to wire up for a Break.

hneemann, 2019-07-18:
> Having Break work in single step would also be useful but not a general substitute. I think there will be situations where the conditions are just too complicated to wire up for a Break.

Hmm! You can allways add a AND gate in front of the Break, which means there is a max delay of only one micro step. And if you use a LUT instead of the AND you are able to define arbitrary conditions with only one micro step delay.

On the other hand: Clocking a circuit to a certain position with the keyboard if very difficult if you have to step over a lot of gates.

dfbg, 2019-07-18:
But less difficult than with a mouse.

Sorry, I wasn't clear enough.  The "f" key would be for the new do-all-the-micro-steps button.  A different key (perhaps overloading "s"?) would be the existing single step button.

hneemann, 2019-07-18:
Please try the latest [pre release built](https://infdigital.dhbw-mosbach.de/).

I have added a "run to break" funktion to the single gate step mode. To make it easier, this button also clocks the circuit if no break was detected unti the circuit has stabelized again. This is my test circuit:

[image]

Let me know how it works at your use case.

dfbg, 2019-07-19:
I reach the cycle limit of the break component every time.  I think it's because I'm dividing the clock and feed several other components with it so the circuit never stabilizes.

When I'm stepping through a clock transition I don't care about I just rapidly click on the step button until the blue ovals all disappear and the button is disabled.  I just want that to happen automatically.  The code already knows when to stop because it's disabling the button.

hneemann, 2019-07-19:
Have you tried to increase the timeout cycle number in the Break component?

dfbg, 2019-07-19:
I have.  It still times out (always at double the number entered).

I don't actually want the clock to automatically run during this fast-forwarding of the micro-steps.  I've been specifically trying to track down a problem that involves two colliding actions (one of which is an asynchronous input) starting during clock cycles that are too close together.  So after checking the state of everything when an input or clock is changed I'll micro-step through it or want to skip to the next transition.

Just having a keyboard shortcut for single-step would speed things up considerably.  Then I could use "C" and "[step shortcut]", hopefully with one hand, to work through things quickly.  Using the mouse changes the focus and "C" doesn't work without clicking back in the main view.

hneemann, 2019-07-19:
Ok! I have reworked the function of the new tool bar button. If there is a break component, the circuit is clocked to the break. This is essentially the same function as in the non-single-step mode, and feels consistent. If there is no break component, only the remaining single steps are executed. 
Both actions now have a hot key "M" and "CTRL+M".
Please try the latest pre release.

dfbg, 2019-07-19:
That works great.  I would only ask that the shortcut for the remaining steps be an unmodified key so I can keep my keyboard hand in one place and use the mouse with my other hand.  I set up my IDEs to use F5-F8 for the common step varieties and it's just automatic now.  I'm not suggesting using those keys - just making them easy to type with a single hand position.

hneemann, 2019-07-20:
What about  [V] (micro step )and [B] (skip mikro steps).
Then the keys  [C],[V],[B] can be typed with one hand.
Or make a better suggestion that does not interfere with the existing assignment. (A list can be found in the documentation)

dfbg, 2019-07-20:
Those keys work nicely.  And even a more comfortable hand position than function keys.

hneemann, 2019-07-20:
Please try the latest pre release.

dfbg, 2019-07-22:
Works great.  Very helpful.

closed 2019-07-22 by hneemann