hneemann/Digital #286: I/O Labels get cropped in exported SVG files

dfbg, 2019-07-19:
When there is a label at the left or right edge of the circuit and it is inverted the text will be cropped because it gets drawn using "¬" and doesn't remeasure the text field for sizing the document.

Something as simple as:
~Input ⊡—⊙ ~Output

The "¬" will be missing on the left and the "t" will be missing on the right.  More of a problem when the outputs are numbered and in a nice column.

hneemann, 2019-07-19:
This is pain! The clean way is to use the attribute style="text-decoration:overline;" in the SVG file.
The problem is that there are many SVG import tools that can't deal with this style attribute, which means that sometimes the line is too low, making the text almost unreadable, or the line is missing, which is a big deal in this case.
What does this file look like in your workflow?

[z.svg.zip](https://github.com/hneemann/Digital/files/3409727/z.svg.zip)

dfbg, 2019-07-19:
I get correct overlines in Safari and then in the preview PDF that I would print.

It also looked correct in Chrome and Mac Finder QuickView.  But not very old Illustrator CS4 (which I tried just because I have it - it's not surprising).

I check off connections on the printouts as I wire the breadboards for real testing. And then again later to confirm the schematics.  I may eventually use them for presentation.

commit 944ca14a (2019-07-19): Uses the correct overline text decoration in svg files, closes #286
closed 2019-07-19 by hneemann