hneemann/Digital #287: +/- shortcut keys don't work in 0.22+

dfbg, 2019-07-19:
The ability to increase and decrease the number of inputs via the keyboard on the basic logic components doesn't work in 0.22 or later.

Works correctly in 0.21.

hneemann, 2019-07-20:
Are you sure! It works fine on my machine, and there is a test case for this feature, so I don't expect it not to work in an earlier version.

dfbg, 2019-07-20:
I just tested each of them again.

It works in 0.21 and 0.22 (I don't know why it didn't earlier).
It doesn't work in 0.22-240.

hneemann, 2019-07-20:
0.22-240 is the head of the new finishStep branch and also there it works fine on my machine.
Are you sure the canvas has the keyboard focus?

dfbg, 2019-07-20:
It definitely has focus as I just clicked on the component to select it.  I also confirmed it with "R" and "L".

I just open a new circuit window.  select an AND gate from the menu and place it.  Click on it and press the keys.  Nothing happens.

hneemann, 2019-07-20:
That's funny! The [+], [-] hotkeys only work, if the component is not selected.

commit 00cae2ff (2019-07-20): fixes #287
closed 2019-07-20 by hneemann