hneemann/Digital #289: Please make git commit detection optional

yurivict, 2019-07-22:
Packages aren't built from the git repository, they are typically built from the source tarball, and they would all fail for this reason.

hneemann, 2019-07-22:
I don't want binaries that don't contain a version number to be flying around. How can a unique version number be ensured in this case?

yurivict, 2019-07-22:
Github doesn't provide .git directory in source tarballs.
The version should be supplied through maven arguments or environment.

hneemann, 2019-07-22:
How is it ensured that the version number set actually matches the version number of the source code used?

yurivict, 2019-07-22:
Build system should ensure this.
Alternatively, you can always commit the version number when you make a release.

hneemann, 2019-07-22:
> Build system should ensure this.

The currently used build system ensures this very reliably! But you're just asking to change that. The question is, how does your build system ensure this?

> Alternatively, you can always commit the version number when you make a release.

I don't want to do that, because like you say: The build system should ensure this.

yurivict, 2019-07-22:
> The currently used build system ensures this very reliably!

What is the current build system? If you mean 'git clone' - it doesn't work in the context of package building, because they use tarballs. `git clone` is also insecure because it doesn't guarantee that it would always produce the same result, and tarballs do.

hneemann, 2019-07-22:
Packages are not important to me. What is important to me is a reliable version number in the distributed binary. If a user has a problem, I want to be able to reliably determine which source code was used to create the binary that that user uses.

The current pom.xml ensures this: When creating a jar, the correct git version number is included. If I change this, the correct version number must be provided by something else.

yurivict, 2019-07-22:
Users install all software from packages. Like ```apt-get install digital```. Building by hand is not an option for most people these days.

hneemann, 2019-07-22:
A more specific question:
When I introduce a maven argument that sets the source code version, how do you ensure that the correct number is always set when you build a package?

yurivict, 2019-07-22:
The port would pass the correct version value.

hneemann, 2019-07-22:
The port?

yurivict, 2019-07-22:
The port that the package is built from.

hneemann, 2019-07-22:
And where does the port get the right number from?
I just want to understand the process.

yurivict, 2019-07-22:
The port has the current version number set as a variable.
The package infrastructure takes this number, and downloads the tarball corresponding to this version number.
Then it builds the package, and can pass the same number to the build, so that this number is always correct.
I am not aware of any cases when there was a mismatch in the version number, declared vs. actual version.

hneemann, 2019-07-22:
Now its possible to pass in the current revision number:

`
mvn install -P no-git-rev -Dgit.commit.id.describe=[num-provided-by-port]
`

commit ebc913a3 (2019-07-22): makes the auto git rev detection optional, closes #289
closed 2019-07-22 by hneemann