hneemann/Digital #292: ROM isnt sending out 10 bits when it has 10 bit values

naltronix, 2019-07-27:
I have a rom that as it says, will not output 10 bit values when it has said 10 bit values inside of itself when the address is for example 0x3-0x6
Contents of ROM
https://gyazo.com/47b6effcb0d633244bed18fb03fee13f

[ALUfiles.zip](https://github.com/hneemann/Digital/files/3438592/ALUfiles.zip)
File for my ALU
If this is an issue with the ALU itself, feel free to delete this

hneemann, 2019-07-27:
I tested your ROM. All stored values are output correctly.
The nine bit values 0x104, 0x108, 0x110 as well as the 10 bit value 0x200. 
I'm not sure what's going on.

naltronix, 2019-07-27:
When it enters the splitter, the outputs for bit 9 and bit 10 are not output and I don't understand if its an issue on my end or if its an issue with the ALU or digital itself

naltronix, 2019-07-27:
It was an issue with the ALU, I am so sorry for bothering you

closed 2019-07-27 by naltronix

hneemann, 2019-07-28:
> I am so sorry for bothering you.

Never mind! That's fine!