hneemann/Digital #294: what is the data width of the Keyboard?

ProxyPlayerHD, 2019-08-03:
It has no options to set the amount of bits it uses for the output, nor is it mentioned anywhere in the documentation.

but i know that it's not 8 bits because otherwise my CPU would work with it... but it's not 7 bits either, so it's not even using ASCII at all?

why is this thing so strange

hneemann, 2019-08-03:
The keyboard has a 16 bit output.

> on a completely unrelated note... why are hexadecimal numbers 0-9 not shown with the hex prefix? it's really weird especially with stuff counting up/down when it's not a consistent size because of the dis/reappearing prefix. same with leading 0s. they would be useful in a lot of cases.

It is possible to select the hex output format for inputs and outputs

> on another note, i cannot seem to generate Verilog, i'm getting the "no verilog code for RS_FF" which makes little sense, i only have Flip Flops in the circuit that don't have async set/reset so there is no reason why it shouldn't compile since a clock based RS Latch is one of the easiest circuit to make in Verilog

What should happen if R and S are both high and a clock pulse occurs? What happens at a RS-FF build from two NOR gates and two AND gates this case?

ProxyPlayerHD, 2019-08-04:
>The keyboard has a 16 bit output.

but why? the Terminal is 8 bit and so is ASCII... it doesn't seem logical...

>It is possible to select the hex output format for inputs and outputs

but is it possible to select that for measurement values? since i'll be looking at that quite often. (plus there are some  things inside the list that i never put as measurement value, like the clock input and some buttons, how do you remove things from that list?)

>What should happen if R and S are both high and a clock pulse occurs?

simple, Reset takes priority. that's what happend with most RS implementations in logic simulators/FPGAs

if Reset is enabled it always sets the output to 0, if Reset is not enabled, THEN it checks if set is enabled, if it is it sets the output to 1.

hneemann, 2019-08-04:
> but why? the Terminal is 8 bit and so is ASCII... it doesn't seem to make sense? what are the extra bits used for?

Because Java uses 16 bit unicode characters.

> plus there are some things inside the list that i never put as measurement value, like the clock input and some buttons, how do you remove things from that list?

There is a menu option which allows you to modify the measurement values.

ProxyPlayerHD, 2019-08-04:
>Because Java uses 16 bit unicode characters.

but then why not unify both the terminal and keyboard to atleast use the same bit width?

>There is a menu option which allows you to modify the measurement values.

and where?

hneemann, 2019-08-04:
> but then why not unify both the terminal and keyboard to atleast use the same bit width?

The terminal has an arbitrary bit size. It is possible to use 7, 8 or 16 bits.
But this is only possible at input signals.

ProxyPlayerHD, 2019-08-04:
>The terminal has an arbitrary bit size. It is possible to use 7, 8 or 16 bits.
But this is only possible at input signals.

oh i see, so i guess i just have to use a splitter to cut off half the bits from the keyboard, since UniCode is backwards compatible with ASCII.

also i couldn't find an option to remove measurement values from the list, i could only find one to order them differently

hneemann, 2019-08-06:
> also i couldn't find an option to remove measurement values from the list, i could only find one to order them differently

You are right. Inputs, outputs and measurement values are always part of the graph. Up to now it is not possible to remove them. I have to think about it.

closed 2019-08-09 by hneemann