hneemann/Digital #297: Request: hover highlighting of net wires

dfbg, 2019-08-08:
I have some circuits with long wires that cross and run parallel and it takes extra focus to make sure I follow the wires and find all the end points.

Having the wires light up, including all tunnel connections, when hovered over would simplify this.  It could be modal - either a menu item (with shortcut) or a modifier key (shift) - so working on a circuit isn't constant color flashing.

hneemann, 2019-08-10:
Please try the latest [pre release built](https://infdigital.dhbw-mosbach.de/).
Unfortunately the Swing ToolTipManager does not allow to register an event listener. Therefore I can't make the highlighted wire disappear so easily. Currently the highlighted wire only disappears when the mouse is moving.

dfbg, 2019-08-11:
I think it's good that the highlight sticks until the mouse is moved away.  Sometimes I need to study the connections.

The line selection for highlighting seems strange.  There's a hover region that extends on both sides of each line and depending on the scale it's not the same distance on each side when there are other nearby lines.  And it appears that horizontal lines have priority over vertical lines.  So if you have a double crossing of double lines (like #) there's only a few pixels where the vertical lines will highlight even when sitting right on top of them.

I don't know if the tool tip api allows for adjusting the hover region.  maybe it's easier to loop through the segment array and do some simple math to find a hit.

hneemann, 2019-08-11:
Please try the latest pre-release version. Now hotizontal and vertical lines should behave symmetrically.
The "catching distance" is always half the grid distance.

dfbg, 2019-08-11:
That works nicely.

However, I think there's something inconsistent with the catching distance around junction dots.  It is possible to have a wire highlight and then move the cursor onto the dot away from the wires (e.g. the top of the circle away from the T intersection of wires) and have it not highlight.  This is even possible by first pausing above the dot but within the catching distance and then moving onto the top curve of the dot.

hneemann, 2019-08-11:
The dots are completely ignored. A wire is an infinitely thin line between the two endpoints.
And I don't think it's worth changing that and paying special attention to the dots.

dfbg, 2019-08-11:
OK.  

It looks like it happens when the cursor is exactly on the coordinate of the wire that stops but beyond the junction.

It's easy enough to just move the cursor.  But it's also easy to discover it.  And it feels like a dot is a natural target to pause over for this because you want to see where the alternatives go.

IvyBit, 2019-08-11:
I'm really struggling generating netlists from a bunch of tunnels and was just about to ask for this feature :-)

hneemann, 2019-08-11:
@dfbg 
I looked at the code again. There was actually still a strange behavior at the ends of the wires.
The latest pre release has fixed it.

dfbg, 2019-08-12:
It's now working predictably for me.  Thanks for looking.

commit 87a78f23 (2019-08-12): makes the wire tool tips optional, closes #297
closed 2019-08-12 by hneemann