hneemann/Digital #299: App crash while editing rom

Johnlon, 2019-08-11:
No other info ....

{code}
java.lang.NullPointerException
        at de.neemann.digital.gui.components.DataEditor$MyLongRenderer.setValue(DataEditor.java:343)
        at javax.swing.table.DefaultTableCellRenderer.getTableCellRendererComponent(DefaultTableCellRenderer.java:257)
        at javax.swing.JTable$AccessibleJTable.getAccessibleChild(JTable.java:7037)
        at javax.swing.JTable$AccessibleJTable.getAccessibleAt(JTable.java:7424)
        at javax.swing.JTable$AccessibleJTable.valueChanged(JTable.java:6939)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:184)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:164)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:211)
        at javax.swing.DefaultListSelectionModel.changeSelection(DefaultListSelectionModel.java:405)
        at javax.swing.DefaultListSelectionModel.changeSelection(DefaultListSelectionModel.java:415)
        at javax.swing.DefaultListSelectionModel.setSelectionInterval(DefaultListSelectionModel.java:459)
        at javax.swing.JTable.changeSelectionModel(JTable.java:2379)
        at javax.swing.JTable.changeSelection(JTable.java:2461)
        at javax.swing.plaf.basic.BasicTableUI$Handler.mouseDragged(BasicTableUI.java:1253)
        at java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:319)
        at java.awt.Component.processMouseMotionEvent(Component.java:6587)
        at javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3342)
        at java.awt.Component.processEvent(Component.java:6308)
        at java.awt.Container.processEvent(Container.java:2239)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2297)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
        at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4552)
        at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
        at java.awt.Container.dispatchEventImpl(Container.java:2283)
        at java.awt.Window.dispatchEventImpl(Window.java:2746)
        at java.awt.Component.dispatchEvent(Component.java:4711)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
        at java.awt.EventQueue$4.run(EventQueue.java:733)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
        at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:715)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
java.lang.NullPointerException
        at de.neemann.digital.gui.components.DataEditor$MyLongRenderer.setValue(DataEditor.java:343)
        at javax.swing.table.DefaultTableCellRenderer.getTableCellRendererComponent(DefaultTableCellRenderer.java:257)
        at javax.swing.JTable$AccessibleJTable.getAccessibleChild(JTable.java:7037)
        at javax.swing.JTable$AccessibleJTable.getAccessibleAt(JTable.java:7424)
        at javax.swing.JTable$AccessibleJTable.valueChanged(JTable.java:6939)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:184)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectionModel.java:154)
        at javax.swing.DefaultListSelectionModel.setValueIsAdjusting(DefaultListSelectionModel.java:685)
        at javax.swing.plaf.basic.BasicTableUI$Handler.setValueIsAdjusting(BasicTableUI.java:953)
        at javax.swing.plaf.basic.BasicTableUI$Handler.mouseReleased(BasicTableUI.java:1166)
        at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:290)
        at java.awt.Component.processMouseEvent(Component.java:6539)
        at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
        at java.awt.Component.processEvent(Component.java:6304)
        at java.awt.Container.processEvent(Container.java:2239)
        at java.awt.Component.dispatchEventImpl(Component.java:4889)
        at java.awt.Container.dispatchEventImpl(Container.java:2297)
        at java.awt.Component.dispatchEvent(Component.java:4711)
[... 101 lines cut ...]
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
        at java.awt.EventQueue$4.run(EventQueue.java:733)
        at java.awt.EventQueue$4.run(EventQueue.java:731)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
        at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:715)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
java.lang.NullPointerException
        at de.neemann.digital.gui.components.DataEditor$MyLongRenderer.setValue(DataEditor.java:343)
        at javax.swing.table.DefaultTableCellRenderer.getTableCellRendererComponent(DefaultTableCellRenderer.java:257)
        at javax.swing.JTable$AccessibleJTable.getAccessibleChild(JTable.java:7037)
        at javax.swing.JTable$AccessibleJTable.getAccessibleAt(JTable.java:7424)
        at javax.swing.JTable$AccessibleJTable.valueChanged(JTable.java:6939)
        at javax.swing.DefaultListSelectionModel.fireValueChanged(DefaultListSelectio
{code}

hneemann, 2019-08-11:
Which version you are using? Can be found in the help dialog.

hneemann, 2019-08-11:
Do you know how to reproduce this bug?
The table tries to render a null value, but the table model never returns a null value.

Johnlon, 2019-08-11:
Latest download today from GitHub

Johnlon, 2019-08-11:
Not sure how to reproduce.

I had been in the room editor.

Btw I noticed the rom editor allows one to enter more digits than the width
of the device.

Perhaps that's related.

Johnlon, 2019-08-11:
Btw is there any way to paste a seq of values into the rom using ctrl-v?

This is possible on logism and is very handy.
Digital seems to require entering on a byte by byte basis.

hneemann, 2019-08-11:
> Btw is there any way to paste a seq of values into the rom using ctrl-v?

No. I think most people write the data to a file and then just read the file.

commit 350916d1 (2019-08-11): fixes #299
closed 2019-08-11 by hneemann