hneemann/Digital #300: Scroll bars stay visible

dfbg, 2019-08-11:
In all the 0.23 pre-release versions I've used the scroll bars appear unexpectedly and stay visible even when not needed.

An easy trigger is to size the circuit with F1 and then zoom one click either direction with the mouse wheel.  The scroll bars show up whether you went bigger or smaller.  When you zoom the opposite direction back to the size that fills the window the scroll bars will still be visible and show that there's a buffer of empty space in all directions.

hneemann, 2019-08-11:
There are two modes possible at the canvas:
In the first mode, the circuit is resized on the canvas and when you resize the window, the circuit also resizes. In this mode, for obvious reasons, there are no scrollbars at all.
In the other mode the circuit keeps its size even if the size of the window is changed. This mode is activated when the circuit has been edited to prevent the scaling from changing when the circuit is modified or if the scaling is modified manually. 
In this mode the scrollbars appear when there is not enough space and a small reserve (20%) is allowed so that components can be placed at the edges of the circuit.
What you observe are these 20% at the edges.

dfbg, 2019-08-11:
I have ICs displayed with the Simple shape along the bottom of some circuits and this behavior covers the part name/number at least.  It can also cover the last character of output labels which might be an index value.

I just find I have to press F1 a lot more now and most of the time it's not changing the zoom.

Can the scroll bars be made a setting?  I had no problems with the drag canvas model.

hneemann, 2019-08-11:
I don't want to make the scrollbars optional but maybe it would help to modify the auto-fit (F1) to give more space around the circuit.

dfbg, 2019-08-12:
Is it possible to have the scroll bars always visible and disable them when not needed?  That way the work area won't shrink when they activate.

hneemann, 2019-08-12:
But that's not the usual behavior. I think it's better to leave space at the auto-fit for the scrollbars.

commit cbe8af91 (2019-08-12): increased the free space at the edges at auto-fit, closes #300
closed 2019-08-12 by hneemann