hneemann/Digital #301: Sub-circuit label rendering

dfbg, 2019-08-12:
The 74682 has active low outputs that are labeled "~P>Q" and "~P=Q".  These render with the overline only over the "P".  This is incorrect as a logic statement.

I've tried many variants and can't get an overline over the symbols.  "~P~=~Q" draws the overline over the P and Q but not the equals.  Closer but also not correct logically.

If there's a reason to not have ~ draw the overline over everything then perhaps something like "~[P=Q]".  Or make ~ into a toggle so "~P~=Q" would produce the current result.

hneemann, 2019-08-12:
Try "~{P=Q}"

dfbg, 2019-08-12:
Yeah, that works.  And now that you mention it I feel like I discovered that once before and forgot (and didn't even think to try it).  Is it in the documentation?

commit 9e0ca16a (2019-08-12): makes the generic shape a bit higher if there is more than one output, closes #301
closed 2019-08-12 by hneemann