hneemann/Digital #302: Mac shortcut keys

dfbg, 2019-08-12:
Is it possible to switch the Attributes pop-up windows to use the command key instead of control for copy/paste in text boxes?

Having one modifier for the main window and the other for the text boxes is frustrating.  Muscle memory is a powerful thing.

keller91711, 2019-08-21:
Agreed. Also double-clicking a component to open attributes would be very useful and intuitive.

hneemann, 2019-08-21:
@dfbg 
@keller91711 
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) v0.22-355-g6b2b6a7
I don't use a Mac so i can't check the behavior.

keller91711, 2019-08-21:
I tried v0.22-355-g6b2b6a but the command- keys still don't work in the attribute dialogs.
Will it be possible to turn the grid off and on?
I didn't see how to go beyond the first step of the offered Tutorial.

hneemann, 2019-08-21:
@keller91711 
> Will it be possible to turn the grid off and on?

In the settings you are able to turn off the grid.

> I didn't see how to go beyond the first step of the offered Tutorial.

What happens if you insert the two inputs?

keller91711, 2019-08-21:
Never mind, I see. I had jumped ahead.
Thanks.
(BTW the red and blue dots are not very visible, at least to my eyes.)

hneemann, 2019-08-21:
@keller91711 
For me it is very important to know if something does not work with the tutorial! In the future it will be offered to every new user. It wouldn't be good if it didn't work properly.

keller91711, 2019-08-21:
Is it possible to back up the tutorial if one goes astray?

hneemann, 2019-08-21:
@keller91711 
In your home folder exists a file named ".digital.cfg".
Setting the key "showTutorial" to true,  enables the tutorial again.

hneemann, 2019-08-24:
@dfbg
@keller91711
Could you please try [v0.23-13-ge3342b0](https://infdigital.dhbw-mosbach.de/) on a Mac.

keller91711, 2019-08-24:
So far it seems good, at least as good as previous versions.

In one week, I will be asking my class to download, so having a stable
version by then would be great.

Did you know that Examples/Combinatorial/4x1-AND.dig is empty? Consequently
the 4x4-Multiplier.dig doesn't work.

Thanks!

Bob

hneemann, 2019-08-24:
> So far it seems good, at least as good as previous versions.

There was no previous version with a working [Command]+[Key] instead of [CTRL]+[Key] in the text boxes.

> Did you know that Examples/Combinatorial/4x1-AND.dig is empty? Consequently the 4x4-Multiplier.dig doesn't work. 

There are no such circuits in the examples/combinatorial folder. And the 4x4 multiplier contains test cases: It is not possible to build Digital if a test case fails.
Maybe you created these circuits yourself?

keller91711, 2019-08-24:
I probably did and forgot. Very sorry.

I’ll check the other thing.

Bob

dfbg, 2019-08-24:
Seems to be working.  Tried it in various Attribute windows and the Test edit window.

Any chance of getting Undo in these windows?  It's easy enough to cancel the Attribute windows but the Test editor just keeps all changes.  An accidental paste of some unrelated text takes manual work to remove.

keller91711, 2019-08-24:
Text-box editing seems to work with command-.

Thanks!

Bob

hneemann, 2019-08-24:
What about [Command]+[Key] doing copy&paste of circuit components?

keller91711, 2019-08-24:
command-c and command-v work

command-x works on a group selection, but not on just a single component
selection.

hneemann, 2019-08-25:
closed in c20fd92b6bb26f9a0a2a25a33b8bea1932f3f270

closed 2019-08-25 by hneemann