hneemann/Digital #307: Too hard to select small components

dfbg, 2019-08-12:
Small components like the diodes and "Constant value" are almost impossible to select.  Even at high zoom levels there's only a few pixels that will select the component instead of starting a new wire.

hneemann, 2019-08-12:
You are able to select small components by using CTRL+Click.

dfbg, 2019-08-12:
OK.

When I ctrl+click on a Constant I have to click right on the wire dot.  But when I right-click I can also click on the value.  Can ctrl+click also support the expanded area?

What happens on a Mac when it has a single button?  ctrl+click is usually the right-click replacement.

hneemann, 2019-08-12:
Digital uses two clicks, and a modifier for each click.

Windows: 

| Type         | Action |
|-------------|------------- |
| Primary click  | Left  |
| Primary click modified  | Left+CTRL |
| Secondary click  | Right  |
| Secondary click modified  | Right+CTRL |

Mac:

| Type         | Action |
|-------------|------------- |
| Primary click  | Left  |
| Primary click modified  | Left+SHIFT |
| Secondary click  | Left+CTRL  |
| Secondary click modified  | Left+CTRL+SHIFT |

To select a small component you have to use the "Primary click modified".

dfbg, 2019-08-12:
That makes sense.  I don't think I've ever had to use/contemplate the ctrl-click model.  I possibly had a two button mouse before the context menu was a system feature.

hneemann, 2019-08-12:
It is possible to enable the Windows mouse mode on a Mac, if a two button Mouse is available.
Maybe that is an option.

dfbg, 2019-08-12:
Yes, I've had that enabled ever since it became an option.  I was just wondering about the one button mouse model - not suffering from it ;-)

closed 2019-08-12 by hneemann