hneemann/Digital #308: Ignored mouse click after group select

dfbg, 2019-08-12:
Drag to select any number of objects.
Right click which will either cancel the selection or open the attributes window.
Cancel the attributes window if it opened.
The next click on empty canvas will not start a wire.

hneemann, 2019-08-12:
That's a problem I've been thinking about for a long time:
How do you recognize when you click into a component that does not have the input focus? In other words, how do you filter the clicks that are made to bring the focus back?
With Swing it seems to be very difficult. I haven't found a really good solution yet. The solution I'm using now remembers when the focus is lost, but doesn't recognize that the focus has returned before the mouse click.

dfbg, 2019-08-12:
It seems to actually only be a problem when the attribute window is presented.  If I only select wires right-click cancels the selection properly.  It also works correctly if I left-click outside the selection rectangle to cancel the selection.

After the attribute window is closed the selection is cancelled.  Can't the focus state be reset in the same place?

hneemann, 2019-08-12:
That's a good idea! It doesn't really solve the problem, but it improves the current workaround.

commit a62ae4f4 (2019-08-12): closes #308
closed 2019-08-12 by hneemann