hneemann/Digital #309: Updates for 74247 and 74248

dfbg, 2019-08-17:
Figured out how to get the ~BI/~RBO pin to work bidirectionally.

Also fixed the output pin labeling on the '247.

I think I added enough new tests to cover things.  Does an "x" automatically test all possible values or does it just use the value from the previous test?

[74247+74248.zip](https://github.com/hneemann/Digital/files/3511523/74247%2B74248.zip)

hneemann, 2019-08-17:
Your solution is neat, but very expensive to simulate.
I have already added the 74247+74248 chips to the library with a simpler although not perfect solution. It does not allow ~RBI low and ~BI/~RBO forced to high at the same time.

dfbg, 2019-08-17:
That seems to be the design model of the real part.  While it won't break if ~RBI is low and ~BI is high there isn't a sensible use for that combination as ~BI will override ~RBI and the ripple blanking won't work.

It seems like if you wanted a ripple blanking chain and to use ~BI to control brightness via PWM you'd have to use an Open Collector output to drive ~BI low.  The Z state of the driver would allow the ~RBO output to assert itself.

Here are new versions with your FET section and my pin and test updates.

[74247+74248.zip](https://github.com/hneemann/Digital/files/3512425/74247%2B74248.zip)

hneemann, 2019-08-17:
> It seems like if you wanted a ripple blanking chain and to use BI to control brightness via PWM you'd have to use an Open Collector output to drive BI low. 

And even worse: I think you need a separate open-collector transistor for each display.

commit 2118dbd1 (2019-08-18): Updates 74247 and 74248, closes #309
closed 2019-08-18 by hneemann