hneemann/Digital #310: Request: Black and White mode SVG export

dfbg, 2019-08-17:
I print the SVG files on a laser printer for breadboard assembly.  The grey text used for most of the labels is barely legible because the printer is using half-tones to approximate the color and is extra bad against the yellow of the typical IC component in Simple mode.

The I/O labels are black and very sharp and it would help immensely if the other labels were as readable.

Many of my circuits create files that are too complex to reasonably try post processing in Inkscape.

hneemann, 2019-08-17:
Are you familiar with LaTeX? The SVG+LaTeX export already does that.

dfbg, 2019-08-17:
I just printed my circuit with the new settings based system.  Everything is now sharp and it's so much easier to read.

hneemann, 2019-08-17:
While thinking about this ticket, I realized that almost every time I create an SVG file, I change something in the code.
It seemed time to clean up the mess. :smile:

hneemann, 2019-08-23:
closed in 8a0b7911ad254a44ddc57d6690428f1505bfe0b4

closed 2019-08-23 by hneemann