hneemann/Digital #313: Suggest not making drawing a wire the default

keller91711, 2019-08-21:
In my humble opinion, it would be better to have drawing a wire require that a key be held, rather than have it be the default action when the mouse moves. I spend 40% of the editing time undoing wire draws that I didn't want.

hneemann, 2019-08-21:
see also #282

hneemann, 2019-08-21:
Which version are you using?

keller91711, 2019-08-21:
I was using

Digital.jar
Build git-Revision: v0.22
Build Time: 2019-04-01 07:40

hneemann, 2019-08-21:
The latest pre release should better avoid a unintended wire drawing.

Can you remember what you actually wanted to do when you clicked into the circuit and a wire was drawn surprisingly?

keller91711, 2019-08-21:
Sometimes it seems like just breathing (just kidding).
Sometimes moving the mouse into the window from another window seems like
it starts drawing.
Another place is leaving a wire just completed at a terminal. This could be
alleviated if one needed to hold a key to draw.

I'll try the latest.

hneemann, 2019-08-21:
> Another place is leaving a wire just completed at a terminal.

If you connect a wire to a terminal, the wire drawing should already be stopped.

keller91711, 2019-08-21:
It seems better in the latest pre-release.

hneemann, 2019-08-22:
@keller91711 
In the past, users have sometimes complained that wires are drawn surprisingly.
But wires are only drawn when you click on the circuit. And that works reliably.
But that means that users click on the circuit, but have something completely different in mind, and are then surprised by the wire. One reason could be that the user only wants to bring the window to the front. That should be fixed.
It doesn't happen to me. So it's interesting for me to know why users click into the circuit when they don't want to draw a wire.
Hence the question if you remember why you actually clicked into the circuit.

dfbg, 2019-08-24:
I think the only time I was getting unwanted lines started was when clicking for focus.  Both from another app and even from the detached Test window.  I just developed the habit of right-clicking when that happened.  It seems to have stopped now in 0.23-13 (I haven't checked other pre-releases).

Requiring a key would be tedious.  I am able to make lots of progress just with the mouse and an occasional 'r' to rotate a component appropriately.  I would rather have a modal scheme using a toolbar (like Photoshop) but I think it would only have two modes so that seems unnecessary.

And what would a mouse click do if it wasn't starting a wire?  If you clicked on a wire would it select it?  That would lead to lots of accidental dragging of existing wires which I think is worse than starting a wire.  Especially because right-clicking when moving a wire just releases the wire where it is instead of reverting it to it's original location.  There would be a lot more Undo-ing.

keller91711, 2019-08-24:
One thing I don't know how to do is remove a wire segment when there is no
convenient end-point, or when it is only connected to other segments. In
the latter case, click at a junction selects both segments, which is not
what is wanted.

I usually end up creating a "stub" in the middle of the wire to create a
new junction, then click that junction.

Maybe there is a better way.

Bob

dfbg, 2019-08-24:
Ctrl-Click on the wire.  I only learned about that recently.  I know your pain.

keller91711, 2019-08-24:
Thanks, but that doesn't seem to work on Mac.

Bob

dfbg, 2019-08-24:
That's what I'm running.  I have the "Use MacOS mouse clicks" setting un-checked.

It's shift-click if the setting is checked.

keller91711, 2019-08-24:
Okay, that works. Thanks for the help.

Bob

hneemann, 2019-08-25:
It seems to me that this ticket can be closed.
If anyone has any thoughts about this issue, I would be happy if it could be added.

closed 2019-08-25 by hneemann