hneemann/Digital #316: Suggestion: Drag on empty background when editing should move the whole circuit, just as it does during simulation.

keller91711, 2019-08-21:


keller91711, 2019-08-21:
In other words, one should not have to control-drag in one situation but not the other. Since drag on an empty background currently has no effect during editing, it should be simple to make the behavior uniform for both modes.

hneemann, 2019-08-21:
Up to now a simple drag shows the select rectangle. Because the select makes no sense during simulation, a simple drag acts like control-drag in this case. 
The intended way is to use control-drag to move the circuit, which works in both, simulation and editing mode.

keller91711, 2019-08-21:
Yes, I see. I was thinking of drag when not around a component, but I agree that it doesn't make sense. 

So I withdraw this.

closed 2019-08-21 by keller91711