hneemann/Digital #317: Option to disable named inputs in the measurement graph

IvyBit, 2019-08-21:
Like all my "problems" this is just a convenience feature... So feel free to close the issue.
I'm simulating a small 64 pin cpld and all of the named  inputs show up in the graph.
Would it be possible to add a "Show in graph" option? 

Thanks for all the effort you put into this!

hneemann, 2019-08-21:
see also #294

hneemann, 2019-08-21:
You are simulating a 64 pin cpld? That's interesting! 
How did you simulate the programming capabilities of the chip.

IvyBit, 2019-08-22:
Thank you, this helps a lot!

Sorry.. but I have to dissapoint you, what I meant was just the logic stuffed in the cpld :-( 
Uart, ps2, video... everything that would be way easier with modern/dedicated chips.
I envy everyone who has enough patience to model even the smallest cpld in digital AND write the toolchain for a part that will never be made into silicon. 

--Tangent 
I jumped on the Ben Eater wagon and somehow thought playing with discrete logic was like I remembered it. Turns out breadboards and hundreds of DIP ics are more expensive than a single fpga! I imagined the 74 series to be sold by weight in buckets... I was wrong.

I simulate everything I had already built years ago with xilinx cplds on an fpga.
My current workflow is a bit strange... Figure out what the hell I've put on those cplds, copy it to a cyclone and test it, simulate the 74 logic in Digital, ignore datasheets, complain about chips not doing what they are "supposed" to do, read datasheets, throw something together in kicad, order it, hope it will work and still be amazed when it does. 
What started as quick nostalgic project on a breadboard turned into a long term feature creep. 
I love it :-)
--/Tangent

commit 56f070e8 (2019-08-23): allows to disable values in the measurement graph, closes #317
closed 2019-08-22 by IvyBit