hneemann/Digital #319: Count switches as inputs?

keller91711, 2019-08-30:
For pedagogical demo purposes, would it be possible to count simple switches as inputs (closed = 1, open = 0, say)?

hneemann, 2019-08-30:
Could you please explain your use case in more detail?
In which scenario should the switch count as an input?

keller91711, 2019-08-30:
Just demonstrating basic logic functions using switches, e.g
series-parallel combinations. Each switch would be identified with an input
variable. It is primarily for pedagogy. Right now I have to use relays to
identify the inputs, which introduces an extra element that is not needed
to understand the concepts.

It isn’t a big deal if it is too hard to put in. I just thought I would ask.

hneemann, 2019-08-30:
Do you want to analyze the circuit to create a truth table? Or should it work as input in a sub circuit?
In which scenario should the switch behave like an input?

keller91711, 2019-08-30:
The former is adequate. It would not be a sub-circuit.

hneemann, 2019-08-30:
Please try the latest [pre release](v0.23-39-gb878f18).
Since there are also use cases, in which the switch may not be an input, there is a new option "counts as input" at the switches.

keller91711, 2019-08-30:
Yes, that is exactly what I was wanting.
Thanks.
Bob

hneemann, 2019-08-30:
closed in 2ad2a9d722878e902245891fb544b6634d81bca7

closed 2019-08-30 by hneemann