hneemann/Digital #320: LED-Switch Combo?

keller91711, 2019-09-02:
For constructing games and instruments, it would be nice to have a single component that functions both as an LED and a switch. 

Here is a possible specification:
The combo has two bits of state: on-off and enabled-disabled.
The LED shows on-off and is settable by one wire input or by user toggling, the latter only if the combo is enabled. The enabled-disabled is settable by a second wire input, but not by the user.
It is open as to whether the LED also shows that the combo is enabled in some way (such as by shading). 
Of course, this combo can be simulated by an LED together with a button, but I am hoping for a single unit.

hneemann, 2019-09-02:
Such a component is far too special to be included in Digital.
But it can be realized as a [custom component](https://github.com/hneemann/digitalCustomComponents).

hneemann, 2019-09-03:
What about a push button combined with a LED which allows to switch on the led interdependently of the button via a separate input?
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

keller91711, 2019-09-03:
Yes, that could serve the purpose. I understand your point earlier.

Thanks.

Bob

hneemann, 2019-09-03:
closed in fa03810c21bf980d1912ff676d943d8f2c256aed

closed 2019-09-03 by hneemann