hneemann/Digital #322: Octal as Number Format in Probe?

SmallRoomLabs, 2019-09-05:
Even if octal is, to say the least,  a bit archaic I'd have good use for it while playing around trying to get a PDP-8 emulated in Digital.  

I guess it's a 5 minute hack to get it working myself, but then I have to install Maven. ^__^  But maybe there's a docker image suitable for quick and easy java dev... Will look into it, might come in handy some day.

closed 2019-09-05 by hneemann

SmallRoomLabs, 2019-09-05:
That was quick - thanks.   

After looking at the code for a bit and now compared my thoughts with your changes I recon it would have taken me an hour or so to get it all understood and and in place. Then would have been thinking about how to implement tests for it as well....

The original 5 minutes guesstimate was taken up by just watching the `mvn intstall` download and install dependencies.  Holy cr*p, here I though that node.js was horrible with all its dependencies and dependencies on dependencies in 7 layers... ;-)

Well, I've learned something today. And now I can compile the prerelease code with the docker container and run it myself and not wait until the next release. Great.

hneemann, 2019-09-05:
>The original 5 minutes guesstimate was taken up by just watching the mvn intstall download and install dependencies. Holy cr*p, here I though that node.js was horrible with all its dependencies and dependencies on dependencies in 7 layers... ;-)

And the funniest: The Digital code base itself comes only with 3 or 4 dependencies.
The rest belongs to maven itself and the generation of the documentation. :-)

commit 6b18c6b7 (2019-09-05): adds octal as number format, closes #322