hneemann/Digital #323: Larger marking of errors in drawing

SmallRoomLabs, 2019-09-05:
I by accident created a short orphan wire segment on the schematics and of course got an error when I tried to run the simulation.  The wire was hidden in a text field and took me a really long time to find even if it was highlighted in red.

Maybe  it could be possible to draw a larger red circle around the highlighted offending item  so it can be found easier?

[image]

closed 2019-09-06 by hneemann

SmallRoomLabs, 2019-09-06:
Cool!  It works just fine - it's now so much easier to find the problem in a larger schematics.

SmallRoomLabs, 2019-09-06:
Seems like I spoke too soon ;-(

SmallRoomLabs, 2019-09-06:
Hehe...  Going back to the "real" released version is apparently not an option ^__^. Not as long as I have newer stuff in the .dig file.

Will try with the previous pre-release - I guess that should work.

[image]

hneemann, 2019-09-06:
Yes, I am very worried about not breaking an old *.dig file when using a new Digital version.
But of course opening a new file with an older Digital version usually doesn't work.

commit e01e100a (2019-09-06): Larger marking of errors at wires, closes #323