hneemann/Digital #324: Keyboard shortcut issues

SmallRoomLabs, 2019-09-06:
I'll bunch a number of issues together here instead of opening separate issues since they all are related to the keyboard.

1) Enhancement: On MacOS the zooming functions really should be use the command key instead of ctrl.

2) Bug: Ctrl-minus works for zooming out, but neither ctrl-plus nor ctrl-shift-plus works for zooming in

3) Enhancement: It'd be great if you could attach the arrow keys to the scroll-events. This would allow easy navigation around in a document. And if the keys also could be used while drawing a wire it would be a really nice addition.

hneemann, 2019-09-06:
Please try the latest pro release. Zooming should now work with the command key.
But in the past i had some issues with US keyboard layout. So I'm not sure that it's working on a US keyboard.

SmallRoomLabs, 2019-09-06:
I'm at the latest pre-release. And I'm using the US keyboard (wouldn't dream of subjecting self to the horrors of coding with an international keyboard ^__^ )

The command doesn't work at all (tried both macOS mouse-click settings just to be sure)

But!  The ctrl-plus actually works. But only until the first click on the Zoom-to-fit button.  After that the ctrl-plus stops working altogether regardless how I first manually zoom in or out with the gut-buttons.

hneemann, 2019-09-06:
Can you look in your home folder, please?
If there are files similar to "Digital_2019-08-08-24-24_21-47-38.log", please attach them to this issue.

SmallRoomLabs, 2019-09-06:
Sorry... A `sudo find /Users/mats/ | grep Digital_` didn't uncover any files except a bunch of pdf scans of old books.

SmallRoomLabs, 2019-09-06:
After trying zooming *many* more times (restarting between each try) I get the distinct feeling that some uninitialized variables are at play here.

It's al very random.

Every now and then the keyboard zooms doesn't work at all. Maybe one time of ten the command- work, but mostly only the ctrl.  Sometimes the plus work and sometimes the minus and sometimes both work, and sometimes actually neither work directly after a restart.  But it mostly seems to die after clicking the zoom buttons.

hneemann, 2019-09-06:
I assume the [ctrl]+[+] and [ctrl]+[-] keys are working in the circuit pane but not in the menu.
If you click on the "fit circuit" button, the circuit pane looses the focus and [ctrl]+[+] and [ctrl]+[-] are not working. As soon as the circuit gains the focus, it works again.

SmallRoomLabs, 2019-09-06:
Hmm, I don't think so.  I now tried clicking at the zoom-all button and didn't click on the circuit pane after that.  This time **ctrl**-plus AND **command**-minus works, one of each!!! But not the other way round.

Definitely super random.

SmallRoomLabs, 2019-09-06:
Well.  I got curious about my theory about uninitialized variables but in the back of my head I had this nagging feeling that Java can't have those. And well yes.  Wikipedia agrees with that ;-)

`Java does not have uninitialized variables. Fields of classes and objects that do not have an explicit initializer and elements of arrays are automatically initialized with the default value for their type (false for boolean, 0 for all numerical types, null for all reference types).` 

So that's not it then....  But random it is anyways.

hneemann, 2019-09-06:
That's really weird.
Please try [v0.23-52-g26459e1](https://infdigital.dhbw-mosbach.de/)

SmallRoomLabs, 2019-09-06:
Tested.  I might have been wrong about the randomness due to my unorganized testing.  It actually seems like command-minus and ctrl-plus always works. At least it does now after 5 restarts and tries with the v023-52

hneemann, 2019-09-06:
Could you please start v0.23-52-g26459e1 and attach a screenshot of the view menu?

SmallRoomLabs, 2019-09-06:
It looks ok at the menu.  But I still have to do CTRL for the plus/zoom_in to work.

[image]

hneemann, 2019-09-06:
Does the latest distribution (v0.23 without the latest changes) look the same?

SmallRoomLabs, 2019-09-06:
The plain 0.23 looks like this

[image]

The File and Edit still have the "command" symbol for their functions though...

hneemann, 2019-09-06:
That's really weird! What the code does is setting an accelerator key code to the zoom in/out actions. Then Swing renders a key code string representation in the menu items itself. And the correct key codes are obviously set. I have no idea why they are not working correctly.

SmallRoomLabs, 2019-09-06:
The plus key is actually = unshifted on an US keyboard. Maybe that is a part of the problem.  I'll try to remap them to something else and see when happens.

I assume that it is done in `gui/Main.java` at `circuitComponent.getInputMap().put(KeyStroke.getKeyStroke(KeyEvent.VK_EQUALS, KeyEvent.CTRL_DOWN_MASK, false), zoomIn);`

hneemann, 2019-09-07:
That was a good hint! Now I remember this problem: #87

hneemann, 2019-09-07:
in the latest pre-release (v0.23-54-g3997b6c) i have tried to fix this issues in a more general way.
Please try how it works. Also please try the '+'/'-' shortcuts in the circuit  (increasing/decreasing the number of inputs in a basic logic gate).

SmallRoomLabs, 2019-09-07:
Ollrighty!   Now it seems all good.  I can zoom in/out with the command +/- keys on a US keyboarded Mac.  And I can also use the +/- keys for adjusting the number of input of gates.

Hopefully it all still works on a windows/linux machine....  ^__^

SmallRoomLabs, 2019-09-07:
By the way - what is the `mvn`-command to just compile the sources into a jar file without running tests and building the dox?   `mvn compile` dosen't give me a jar, and `mvn package` also runs the tests ;-(

hneemann, 2019-09-07:
This is how to skip the tests:
```
mvn install -Dmaven.test.skip=true
```

hneemann, 2019-09-09:
closed in 80138b7cf90a3cf10c027e2eebc0e9f009318495

closed 2019-09-09 by hneemann