hneemann/Digital #325: Unnecessary circling of objects while editing

SmallRoomLabs, 2019-09-06:
Commit e01e100ae618d8d772d785a1f24855133eb54153 for #323 that circles small error objects unfortunately also applies when selecting items or drawing new wires while zoomed out. This makes it look quite messy.

Selecting:
[image]

Drawing new wire:
[image]

hneemann, 2019-09-06:
That looks really terrible! :-(

closed 2019-09-06 by hneemann

SmallRoomLabs, 2019-09-06:
This is still happening in v0.23-52-g26459e1

[image]

hneemann, 2019-09-06:
I have created the CTRL branch before i fixed this issue. So v0.23-52-g26459e1 does not contain the fix.

commit ceb138ea (2019-09-06): closes #325