hneemann/Digital #326: Attaching a wire "from behind"

SmallRoomLabs, 2019-09-07:
If I have a circuit like the first AND below and want to add an LED to the output like in the second then it seems like the NOT gate gets disconnected  from the AND and the wire goes straight from the LED to the NOT.

To work around that issue I can connect the wire from the LED from the top or bottom into the output of the AND.

Is this "disconnect from output"-behavior intended or is it a bug?

(To be 100% clear - in the second example I first wire up the NOT to the AND, then I wire up the LED to the and in a separate step. I don't just make one wire that just passes by the AND output)

[image]

hneemann, 2019-09-07:
This behavior is intended. A connection is only established to an end of a wire.
But it's possible to connect multiple components to the same wire endpoint.

[image]

SmallRoomLabs, 2019-09-07:
Ah, that makes sense.  So two horizontal wire segments that are joined becomes *one* segment, and thus loses its middle endpoint.

Having the LED directly at the endpoint as in your example definitely looks better than my crooked wire.  I'll change my design to do it that way.

closed 2019-09-07 by SmallRoomLabs